iNavFlight/inav #5721: LiPo Battery Cell Monitor - sensor support

Simmy07, 2020-05-13:
**Introduction**
I would like to propose new feature in order to improve battery capacity estimation along with notifying user about this.

**Is your feature request related to a problem? Please describe.**
User estimate battery usage either using timer, entire pack voltage or pack voltage plus current consumption which gives rough consumed energy. This brings disadvantages like a knowledge when battery is low for 2s, 3s (and so on) or a need to remember which battery pack is currently used in a case of consumed energy. There may be also old batteries, which do not provide full capacitance.
Users also often use cheap cell's monitors which not only does not provide accurate readings, but they do not provide any feedback besides embedded buzzers. Others existing solutions provide only WiFi access to the battery status or are designed to work only with *FrySky* transmitters.

**Describe the solution you'd like**
I am developing a *LiPo Battery Cell Monitor* board which can monitor voltage of each cell independently. While battery voltage is a poor estimate of remaining capacity I believe it is way better than above solution. This board uses comparator to extract exact cell's voltage. Later those readings can be presented via serial port (I2C advised due to UART ports being to precious; SPI would consume ADC). Board could present per-cell voltage, entire pack voltage and (the most important) the weakest cell voltage. The weakest cell voltage is actually the one that could be presented via OSD, telemetry (this for example solves issues with cheap *FlySky FS-i6* transmitters), low-battery warning buzzer, on-board status LED flashing and LED strip patterns to user. In order to save main CPU cycles, *LiPo Battery Cell Monitor* can smooth out readings itself either by using average or some sort of filtering.

[image]

Currently, *LiPo Battery Cell Monitor* is designed for use with 2s-4s battery packs. It is possible to extend it for use with 2s-6s battery packs by introducing one more comparator and redesigning the board. MCU has enough ADC's to support such solution. In a later stages of the development a current sensor could be added along with a voltage SAG calculation. If attaching this sensor permanently to the battery would be needed, there could be added an transistors or an analog switch to minimize further power consumption.

P.S. This is an copy of the similar issue requested at Betaflight site but it seems iNav can be more appropriate fork to implement this.

digitalentity, 2020-05-13:
Not sure if this is a good idea. On a **good** battery cells discharge more or less at the same rate and have more or less identical capacity - therefore monitoring overall voltage is equally as efficient. 

Regarding choice of the bus: reading the ADC, averaging the data on the CPU is cheap, I2C is slow and time-consuming and also prone to errors. I'd recommend using an UART and make it MSP-capable.

Regarding choice of MCU - 5V MCU is a bad idea. Our CPUs are 3.3V devices and many interface pins there are not 5V tolerant, therefore connecting this device is simply not safe. Mixing 3.3V and 5V devices will also cause weird issues on I2C bus. I'd recommend using a device from an STM32F030 line.

shellixyz, 2020-05-14:
@Simmy07 You might be interested by this PR I just made. It allows you to connect any FrSky sensor to the FC like for example a FLVSS which can measure individual battery cells voltage up to 6S.

https://github.com/iNavFlight/inav/pull/5724

closed 2023-03-06 by DzikuVx