iNavFlight/inav #5731: Tramp power level change, intermittent with GF

Jetrell, 2020-05-15:
Testing was done with  - 
 **Matek F411-Wing** 
 **Matek VTX mini**
[inav diff all 2.5.txt](https://github.com/iNavFlight/inav/files/4632802/inav.diff.all.2.5.txt)

iNav 2.5 VTX power level control is intermittent when changing the output level up or down.  It seems like a timing issue.

This is the first time I have used VTX Global functions for power control. I used the `Adjustments Tab` for VTX control and a Logic/Special function in my TX to do the same on 2.3/2.4.

The VTX logic and GF, appear to be working correctly, according to the OSD readout. But the power levels do not always change with the OSD numbers. But it always changes when Arming and Disarming.

I went back to 2.4 FW again and used the `Adjustments Tab` power level/ OpenTX mixing and it worked fine.
I have used a SA2.0 VTX with 2.4 in the same way, and it also worked fine. (But I don't have that VTX anymore).

digitalentity, 2020-05-15:
Does it work as expected when you change the power level from the OSD?

Jetrell, 2020-05-15:
No. When a different power level is selected and confirmed in the CMS. The power level doesn't change. It remains at 25mW, on the OSD read out, and the VTX's 25mW power LED doesn't change either.
But the channel and Band can still be changed from the CMS.

Jetrell, 2020-05-15:
@digitalentity @Scavanger  bit more info. 

I reloaded the development version for SA2.1. My Unify works fully in the CMS.  It will change the power levels too.  But still no joy with Global Function power level changes.
Although the Global Functions will allow the changing of both Bands and Channels via switch for Tramp and Smart audio.

Jetrell, 2020-05-18:
@DzikuVx Any chance of you looking into this, to confirm? 
  Thank you :-)

Strangedave, 2020-05-22:
I am brand new to inav, I have built a lot of betaflight quads. This is my first wing. So far i have not been able to get Smart audio to work.  Running Inav v 2.5 rc1 and a Lumenier LUX Mini 20x20 5.8GHz FPV Video Transmitter (25-800mW). The VTX was working in a quad last week. I did unfortunately murder it all the way, so the spare parts are now living in a wing. It was however working when removed from the carcass. Not sure if is me or the FW. I can see video. just cant change any settings. When I try to change settings its asking me questions I don't know how to answer. Is there a wiki for this?

Jetrell, 2020-05-22:
@Strangedave Hi, The FW for SA 2.1 will not be released until 2.5RC2. At present only Tramp and SA 2.0 or less work with iNav.

This issues is related to iNav not allowing VTX power level control on either VTX protocol, through the Global Functions, via a radio transmitter switch.

Strangedave, 2020-05-22:
Awesome, thanks. I thought I was going crazy.

Strangedave, 2020-06-03:
I updated to 2.5 rc2 last night. Still no osd  vtx control. Is there anything else I need to do for sa 2.1. On Lumenier LUX Mini 20x20 5.8GHz.

Jetrell, 2020-06-03:
@Strangedave How did you test it? With the CMS stick menu via the OSD? 
I haven't tried RC2 with SA2.1 yet. But the development firmware I used that had SA2.1 worked fine for me. 
But the Global Function control didn't. And it would also lock out the stick menu control. So the GF had to be cleared in the CLI for the stick menu to give full VTX control.

Strangedave, 2020-06-04:
I tried setting the band, Chanel, and power in the configurator first. When that didn’t work I powered up with a pack and tried in the osd. So maybe I need to clear everything out then try in the osd.

Strangedave, 2020-06-04:
I saved out diff all, removed all lines that had anything to do with the vtx. re-flashed the fc, and saved the diff back into the cli. put a pack on and tried to set the band and channel in the osd. No luck, I'm going to install a tramp vtx in there and see if that is good. I'm not 100% certain the vtx in there is even ok. It came out of a crashed (completely SPLAT) betaflight freestyle setup.

Strangedave, 2020-06-04:
Just as a side note, the settings in the osd did seem to be correct for this vtx.

Mateyhv, 2020-06-04:
No method works for me either on SA2.1 in iNav 2.5 RC2, it does the same as in iNav 2.4, that is nothing. I shall try pitmode though... @Strangedave you can edit your message to add info so that you don't flood the thread.

Jetrell, 2020-06-06:
@Strangedave @Mateyhv 
I loaded 2.5RC2 on my F411-Wing and using a SA 2.1 `Unify Pro(32) 5G8 Race V2 `. I can still get it to work with the CMS. It will change Channels, Bands and Power levels ....But the changes DO NOT appear on the OSD power level and readout. It remains at  power level 1.
Even with low power disarm enabled, it disregards it.

I know SA2.1 is working, because I have a 5k resistor between the Data and Audio pins on the VTX. So I can hear the audio data pulses being sent. Plus I'm also watching the power level change on a spectrum analyzer. 
But the Global Functions have no effect at all with Smart Audio, even with disarming, While they work some what intermittently with Tramp.

I assume that you guys have Unlocked the VTX's power select mode? Red and Blue LED on together.

Mateyhv, 2020-06-06:
@Jetrell to be honest I am not sure which SA version my TBS ```Unify Pro HV``` uses but I am completely unable to change any parameter via CMS or get any info from the VTX. Matek F411 Wing 2.5RC2 with 2 softserials enabled, SA is connected on SS1_TX and tried even different baud rates to no avail.

DzikuVx, 2020-06-08:
There is a bug that prohibits VTX changes after armed. Will be solved in the next release hopefully

digitalentity, 2020-06-08:
@DzikuVx I think we should allow VTX driver to poll the VTX regardless of armed state, WDYT?

DzikuVx, 2020-06-08:
yes, yes we should

breadoven, 2020-06-09:
I found that an Eachine TX5258 works as expected when disarmed, the power level indication on the Tx and in the OSD display changes as the global function logic controlling it changes. However, when armed the Tx responds only ONCE to the global function logic and the OSD display remains stuck at the disarmed setting. Disarm and all is well again.

DzikuVx, 2020-06-09:
I have tested with the fix from #5823 using Rush Tank VTX and everything seems to be working fine both when Armed and Disarmed. Changing Band Channel and Power is possible and VTX keeps new values locked it.

Also, Adjustments and Global Functions use exactly the same mechanism of modifying data.

closed 2020-06-09 by DzikuVx

Mateyhv, 2020-06-12:
Working now! Tetsted on iNav 2.5.1, SmartAudio 2.0, logical conditions + global functions. Thanks!

Strangedave, 2020-06-13:
I wish I could confirm, I unfortunately killed that setup. If i can get it going again ill test and get back to you

PR #5823 (2020-06-09): [VTX] Always process VTX driver updates, regardless of arming state
Fixes #5731