iNavFlight/inav #5735: Quad starts to yaw fast after take-off

kibinek, 2020-05-15:
## Current Behavior

I have installed INAV 2.5.0 RC1 on my MATEK F722SE. When I arm the quad and start moving throttle forward, motor revolutions increase linearly to Tx stick position. When the quad takes-off, motor revolutions increase rapidly and quad starts to yaw fast.
## Steps to Reproduce

1. wait  for GPS fix
2. arm the quad
3. start moving throttle stick forward until quad takes-off
4. quad starts yaw quickly

## Expected behavior

Quad shall behave "normally", no automatic yaw like hell.
## Suggested solution(s)

No clue.
## Additional context

https://pastebin.com/yXwg9Lx6
---

- FC Board name and vendor:
INAV/MATEKF722SE 2.5.0 May  9 2020 / 15:10:58 (272de567d)
- INAV version string:
GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

stronnag, 2020-05-15:
Do you really want `set motor_direction_inverted = ON` ?

kibinek, 2020-05-17:
> Do you really want `set motor_direction_inverted = ON` ?

My dear, THAT WAS IT!!! Not it works normally. I set motor_direction_inverted = OFF.
I also recognized that this variable was not present in INAV 2.4.1, so I'd **strongly suggest to set this variable to OFF as default!**
I also did not recognize a control switch within INAV Configurator 2.5.0. Or is it somewhere there?

giacomo892, 2020-05-17:
It was a bug with 2.5RC1 configurator.

closed 2020-05-17 by giacomo892