iNavFlight/inav #5738: Simulation / virtual FC

samcollins910, 2020-05-15:
Feature request

Would be possible to build into configurator the ability to create a virtual fc so that a build can be simulated and tweeked without having to have a live build on a live board. With the advancements in logic programing it would allow the ability to experiment as thought the configurator was connect to an Fc without the possibility of changing something and potentially killing a very good set up. I know you can always backup and restore a full dump but I feel if this was built into the configurator it would be a very beneficial feature.

wx4cb, 2020-05-24:
totally agree. would be great to be able to import a diff or a dump to look at your config without having to connect up an FC. This i think should be in the configurator repo though not  the normal one.

DzikuVx, 2020-06-11:
Such a "simulated" flight controller will not be developed by INAV. I suggest investing in a separate board for such experiments.

closed 2020-06-11 by DzikuVx