iNavFlight/inav #5740: Optical Flow Configuration Question

GregMichalski, 2020-05-16:
@digitalentity in your instructions for setting up the Matek 3901-L0X, your instructions state to "tilt it from one side to the other by roll and then by pitch axis" and then compare the 0/2 and 1/3 debug traces for similarity. Can you clarify which pair of traces correspond to roll and which pair corresponds to pitch? In the screenshot provided 1 and 3 show activity first, then 0 and 2 which per the described process would suggest 1 and 3 are roll, however no orientation entered results in that correlation. 

Additionally can you elaborate on the polarity of the traces? Does pitching up result in a positive change on the trace or a negative change? Same question for the roll trace - is roll left a positive or negative change?

pDR8iLQ4go, 2020-05-16:
I‘m interred in the answer as well, because I still have some trouble with this sensor as explained here: https://www.rcgroups.com/forums/showthread.php?3572803-Matek-Lidar-and-INAV

pDR8iLQ4go, 2020-05-19:
@GregMichalski: Does the SURFACE mode work on your copter? Which kind of flight controller do you use?

GregMichalski, 2020-05-21:
> @GregMichalski: Does the SURFACE mode work on your copter? Which type of flight controller do you use?

I'm running a Holybro Kakute F7. At the moment I'm holding on iNav on that build so no plans to test Surface mode anytime soon.

closed 2020-08-08 by stale[bot]