iNavFlight/inav #5741: MANUAL (previously "passthrough") mode overrides POSHOLD. Intended behaviour? (2.4.0).1)

CdtSylvestre, 2020-05-16:
From reading the INAV documentation, one is lead to believe that activating POSHOLD automatically sets other modes accordingly, i.e. that ANGLE and ALTHOLD are activated also. This is the case if the active flight mode is the default acro/rate (AIR mode on OSD) mode or angle mode, but **NOT** if the flight mode is MANUAL.
 
The documentation section on NAV modes states that "_Flightmodes are self contained. For example: with RTH and WP (Waypoints) it's not necessary to enable angle, althold or mag, it enables what it needs._", followed by a table indicating that POSHOLD automatically also sets ANGLE and ALTHOLD.
See https://github.com/iNavFlight/inav/wiki/Navigation-modes

To add to the confusion, launch mode and RTH automatically set ANGLE mode while these are active, even if engaged from MANUAL flight mode.

This may be intended behaviour (thinking of MANUAL as an override in most situations except RTH - which may have been triggered by failsafe - and launch where it is clear that the pilot intends to leave the controls to the autopilot) but can bite an unsuspecting plane pilot who likes to fly mostly on manual mode, but uses launch mode as a convenience and switches to POSHOLD after climbing to put on her FPV googles. The scenario is thus:

1. Prep plane for flying (mostly) in MANUAL flight mode. MANUAL is engaged, then LAUNCH MODE, ARM, throttle up...
2. ... launch, plane climbs in autolaunch mode, pilot takes over (in preset manual mode) and flies to a convenient location for an holding pattern...
3. ... pilot selects POSHOLD to get her hands free for removing her sunglasses and putting on her FPV googles, but unbeknownst to her, the plane is still in manual mode and is thus controlled by no one at that time (it may even have started to fly in what might seem to be a circle by chance, if the wings were not level when the pilot stopped to actively control the plane, thinking that Otto was in the pilot seat while it is not).

Question: is the current behaviour intended?

Suggestion:
- if it is, this should be mentioned clearly as a warning in the docs for MANUAL mode and POSHOLD mode (i.e. switching to MANUAL will not allow to engage POSHOLD).
- if it is not, then this should be filed as a bug; POSHOLD should engage from any flight mode including MANUAL.

breadoven, 2020-05-16:
I've always seen MANUAL as an override of stabilised/automated modes. So with MANUAL deselected you are in stabilised mode with varying degrees of stabilisation/automation from ACRO to WP. The only stabilised/automated mode I've found that overrides MANUAL is RTH as you noted.

CdtSylvestre, 2020-05-17:
Thank you, this completely makes sense indeed. To avoid any possible confusion, I added two short sentences in the Wiki in the MANUAL mode section of the Modes page.

closed 2020-05-17 by CdtSylvestre