iNavFlight/inav #5742: Airspeedsensor callibration feature request

obberator, 2020-05-16:
## Current Behavior
The value for airspeed is over 0 km/h, sometimes about 10km/h with covered pitot tube. In flight there is a risk of getting in a stall because the plane ist moving 10km/h too slow

## Desired Behavior
The value of the airspeed should be manually or automatically set to zero

## Suggested Solution
the airspeed should get more attention over all because it is the real speed value for a fixed wing. If you are traveling with 50 km/h GPS Speed over Ground and get 30 km/h tailwind the plane will fall out of the sky. If you get this in manual control there is no problem, but if you are flying an uav mission .....

Cris-77, 2020-05-22:
In addition to the 'zeroing' feature, it would be useful an airspeed ratio settings option, manually or automatically. Not all sensors have the same factor of bank angle of linear characteristics.

MrMap, 2020-10-08:
In general airspeed sensors are noisy / inaccurate close to zero velocity and pretty reliable in the flyable part of the scale - including close to stall. I think it´s their nature due to relying on differential air pressure measurement.
I see your point but I don´t believe there is a real accuracy problem here. It´s just a bit annoying to see the obvious error when its parked on ground :)

kmieloszyk-luxoft, 2020-10-09:
You have right, even commercial aircraft have airspeed indicator with the scale from 40km/h. In analog speed sensors for RC in ~max speed you have error 0.5km/h,  in ~stall speed you have 2km/h error. But it would be good to have **just as for current sensor** , offset and **scale** factor to manual calibration.

kmieloszyk-luxoft, 2020-10-09:
That's why you should turn on the controller when you assembly or prepare for flight (to warm up) than before the flight restart again to re-run autocalibration. Calibration of scale is other problem, because the result is proportional to sqrt of voltage.
Anyway manula calibration is described here https://ardupilot.org/plane/docs/calibrating-an-airspeed-sensor.html 
If I got on the groud 8-11 km/h is not so noisy but with the wrong offset.

MrMap, 2020-11-25:
I withdraw my statement: "Actually, calibrating close to zero readings might create more problems than it solves."
A final preflight calibration opportunity might be a good idea, should not do any harm.

Also, it would be useful with an iNav general airspeed calibration flight feature where airspeed is compared with Ground Speed during a number of in-air racetracks, preferably done a calm day. ArduPilot has this feature.

michalelektryk, 2021-04-27:
If you could at least set scale and offset from CLI somehow...

closed 2024-02-02 by DzikuVx