iNavFlight/inav #5744: GPS all of a sudden not working home arrow gone and flashing

wingflyer101, 2020-05-17:
using a f405 and bn880 GPS everything is fine and has been working fine apart from my home arrow blinking then when I arm I'm left with a small line that blinks so can't tell were home is I've been ripping my hair out trying to fix this, I have a 3d fix and 15-16 sats. I have tried another gps with exactly the same problem.

AnandNL, 2020-05-21:
during flight or on the ground? i have had in flight that de gps stopped working, the runcam split turned out to be the problem jamming the GPS 

[https://www.youtube.com/watch?v=JpVrqozWXbY](url)

[https://youtu.be/jo6tbFnQrfI?t=36](url)

DzikuVx, 2020-05-27:
the fact that GPS module was able to get the fix does not mean it will be able to keep it in a changing environment: electrical noise, RF nosie, weather conditions, etc.

closed 2020-05-27 by DzikuVx

pandg-boop, 2025-11-14:
Hello. I reopen this thread because I have the same problem.

Inav 8, FC rcflying F4WSE mk 1.5.
I am using it on RC boat, everything works fine except that before arming direction to home indicator is flashing and after arming it turns into a flashing dash. Gps have a solid fix.

sensei-hacker, 2025-11-14:
@pandg-boop  please attach your diff and a photo of your craft showing the mounting location of the GPS and what's around it.

Also, verify you have uploaded / switched to the proper INAV font for whatever video system you're using.

pandg-boop, 2025-11-14:
> [@pandg-boop](https://github.com/pandg-boop) please attach your diff and a photo of your craft showing the mounting location of the GPS and what's around it.
> 
> Also, verify you have uploaded / switched to the proper INAV font for whatever video system you're using.

Hi.

In the photo, the gps is in the bow of the boat, around 11 cm ahead of the flight controller (boat controller) that is below that brown cover, just ahead the mast.

Servos are behind the switch you see on the left, toward the stern, while analog VTX and SBUS rx are on top of the mast, around 75 cm above the hull.

Camera, with servo gimball, is installed, and perfectly working, at the stern.

Exact behaviour is that the return to home arrow flashes continuously since the power up, even after GPS fix and sensors calibrating is complete. 
As I arm the boat it turns into a flashing horizontal dash. I moved it on top of the screen but the behaviour is the same.

I have systems and warning messages on and everything seems to be ok after arming, the only problem is direction to home. Even distance from home is correctly updating while sailing.

Video format is set to "AUTO" (I am using PAL however) and the font I selected is impact mini, but i changed to other fonts without success.

Thanks for the attention.

[INAV_8.0.1_cli_20251114_160319.txt](https://github.com/user-attachments/files/23550074/INAV_8.0.1_cli_20251114_160319.txt)

[image]

pandg-boop, 2025-11-21:
Great!! Now I know where is the problem. I had understood, looking around the net, that compass was not necessary, I understood wrong.

Actually my gps antenna is this one:

https://it.aliexpress.com/item/1005005134658746.html?spm=a2g0o.order_list.order_list_main.72.41ae3696inz5S0&gatewayAdapt=glo2ita 

and I think it does not have the compass. If not, as I believe, I will buy the BN880, as you suggested.

Thank you so much. I'll let you know.