iNavFlight/inav #5746: Sport telemetry broken with hardware inverter: smartport_uart_unidir missing in INAV 2.5.0RC1

donchia, 2020-05-17:
## Steps to Reproduce

1.INAV 2.5.0RC1
2.Issue command set smartport_uart_unidir = ON Telemetry.md
3.Error ### ERROR: Invalid name

## Expected behavior
smartport_uart_unidir is used for "SmartPort (S.Port) with external hardware inverter"  as described in Telemetry.md

Current version
# version
# INAV/MATEKF405 2.5.0 May  9 2020 / 15:05:07 (272de567d)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

stronnag, 2020-05-17:
Please read the release note, this variable has been renamed.

closed 2020-05-17 by stronnag

digitalentity, 2020-05-17:
The setting is replaced with a generic `telemetry_inverted`, the setting is relative to "normal" protocol default. I.e. Smartport uses inverted UART, so setting `set telemetry_inverted = ON` will invert the inverted-by-default UART and result in uninverted UART, which is exactly what you need.