iNavFlight/inav #5747: Add support for Ping RX ADS-B Receiver

digitalentity, 2020-05-17:
ADS-B receivers with a reasonable price are popping up, i.e. this one:
https://www.unmannedtechshop.co.uk/product/ping-rx-ads-b-receiver-for-uavs

Would be awesome to have it supported in INAV to detect nearby aircraft

stronnag, 2020-05-17:
See also https://github.com/stronnag/mwptools/wiki/mwp-Radar-View

stronnag, 2020-05-17:
With the device on the ground, my user was seeing aircraft out to c. 40km. On a quiet (covid-19) day in Florida, around 20-30 aircraft active at any time, total of around 67 over a 40 minute log.

I have some logs ....

FranzAr, 2020-05-17:
This would be great to have, especially for long(er) range flights. OSD integration would be awesome too :-)

skaman82, 2020-05-17:
Some documentation: https://uavionix.com/downloads/integration/uAvionix%20Ping%20Integration%20Guide.pdf

ShikOfTheRa, 2020-05-20:
I have just added into MWOSD. Would be great to add into iNav natively. It's also great way to get "buddy" flight from third party solution.

Bottom left of screen is what is on OSD. The rest is just for my amusement. Just can't test fly it yet due to lockdown..
https://www.youtube.com/watch?v=OnCK6YZma6I

I'll happily test for someone as I have one of these units.

ShikOfTheRa, 2020-05-28:
I have released this for MWOSD users.
I can have a go at adding into iNav in a few weeks if no-one else want to pick this up.

It also makes a fantastic solution for "buddyflight" using LORA type serial modems and sending out emulated ADSB mavlink packet . 

[image]

https://github.com/ShikOfTheRa/scarab-osd/wiki/ADSB

skaman82, 2020-05-28:
@ShikOfTheRa yeas, please! I would be awesome if you could add it to INAV

skaman82, 2020-05-28:
@ShikOfTheRa is there a setting for a minimum alert distance? It would be cool if the alarm is only displayed in the osd if the distance is smaller than the set boundary (like 5km)

skaman82, 2020-05-28:
@ShikOfTheRa Could you point me to the code or project that emulates an ADSB beacon on a LORA module? Thank you!

ShikOfTheRa, 2020-05-28:
@skaman82:

I am quite surprised how much I like having that info on the screen. Those times you hear a plane, but not quite sure how far away it is…

can configure:
max range of ADSB you care about.
ADSB flashes if within preset height / distance.

ADSB send code is here. It is just part of MWOSD:
https://github.com/ShikOfTheRa/scarab-osd/blob/master/MW_OSD/SerialMAVLINK.ino#L174

Diagram in link to aeromax manual below. Section titled “APM / PX4 installation with Buddyflight - FPV partner awareness”
https://github.com/ShikOfTheRa/scarab-osd/blob/master/OTHER/DOCUMENTATION/VIRTUALPILOT%20AEROMAX/VP%20AEROMAX%20manual.pdf

For adding to iNav, I am not familiar with the codebase but sure I could put together the basics. I’ll happily do it if one of the other dev’s isn’t already working on it.

skaman82, 2020-06-29:
@ShikOfTheRa any updates on this?

ShikOfTheRa, 2020-06-30:
I have it working on ardupilot
Also have been working to implement it as a "buddyflight" option for awareness of other FPV fliers. It is working on the bench - pending test flights to confirm.

I have started on inav but it will take me a little time as completely unfamiliar with it. It's probably 50% done

digitalentity, 2020-07-01:
@ShikOfTheRa I'm happy to do the basic support, but it might be tricky as I don't have the hardware to test. If you have the hardware on your hands - it would be awesome if we can collaborate

stronnag, 2020-07-01:
@digitalentity I have multiple logs (donated by an mwp user). It's a broadcast device so that's all I used to add mwp support. You're welcome to them if that helps.

ShikOfTheRa, 2020-07-03:
@digitalentity happy to do any testing.

if not urgent I am happy to put together a working model for critique - should be no more than 2 weeks due to the day job... 
half way there now but work got in the way...

tigert, 2020-07-21:
It would be useful to have an open source receiver one could integrate with all of this. 

@ShikOfTheRa you are aware of the inav radar project (that has not had much updates lately though) that displayed the buddy planes on the OSD at roughly the correct spot on the screen when it would be on field of view https://github.com/mistyk/inavradar-ESP32

ShikOfTheRa, 2020-07-28:
@tigert – no idea what they used, but I’m interested too! 

Yes have seen the radar project. For sure the adsb data could populate the radar structure. Kind of see that as a phase 2. Have ordered a couple of LORA modules to set up a test rig. Not sure I would use it as it is - bit too much data on the screen for me 

I do have a working model with ADSB in iNav now. Seems to be working OK, but need to check the altitudes are accurate and consider what user cli adjustable default exclusion ranges are. I was thinking about anything outside 5k distance being out of interest area. 
[image]

skaman82, 2020-07-28:
@ShikOfTheRa  yes I think 5km is a good range for the notification. Could be also configured in cli

ShikOfTheRa, 2020-10-07:
PR submitted
Wiki page ready describing implementation:
https://github.com/ShikOfTheRa/inav/wiki/ADSB

wx4cb, 2022-02-26:
@digitalentity @ShikOfTheRa any updates on this?

ShikOfTheRa, 2022-03-01:
Hi - I did raise a PR for this but it did not get approved.

f4b0, 2022-03-09:
Hi, 
I found two interesting projects [Open Glider Network](http://wiki.glidernet.org/) and [SoftRF](https://github.com/lyusupov/SoftRF) which are well known in the glider community.
The [hardware needed](http://wiki.glidernet.org/esp32-ogn-tracker) to send/receive position information to/from other aircrafts is the same way as it's used by the [Inav-Radar project](https://github.com/OlivierC-FR/ESP32-INAV-Radar)
There is also a way to[ connect the tracker to a flight controller](http://wiki.glidernet.org/esp32-ogn-tracker#toc5) using Mavlink

mariuszlabedzki, 2023-01-29:
would be great to have this function in iNav. Any chance your PR gets approved?

wx4cb, 2023-01-29:
> @ShikOfTheRa I'm happy to do the basic support, but it might be tricky as I don't have the hardware to test. If you have the hardware on your hands - it would be awesome if we can collaborate

@digitalentity if you want me to send you a device i can as long as i get it back.. it's the same one that generated the logs for @stronnag ... gunna have to try and remember where I put it tho. that or i can provide you with a bunch of raw data.. it's output is basically a form of mavlink

dcan999, 2023-05-28:
Did this ever go anywhere? I'd love to have a PingRX on my plane showing me the closest aircraft, I only run INAV on Matek controllers though.

wx4cb, 2023-05-28:
> Did this ever go anywhere? I'd love to have a PingRX on my plane showing me the closest aircraft, I only run INAV on Matek controllers though.

not as far as inav firmware goes AFAIK, but MWP has native support for it and works great... and to be honest... if you get the module up around 15 feet (ie up with your fpv antenna) then you'll receive upwards of 20 plus miles. which is generally more than you need.

https://github.com/stronnag/mwptools/wiki/mwp-Radar-View

dcan999, 2023-05-28:
> not as far as inav firmware goes AFAIK, but MWP has native support for it and works great... and to be honest... if you get the module up around 15 feet (ie up with your fpv antenna) then you'll [...]

I am new to github in general, this link makes it seem like it was working with INAV? Just going by it saying the PingRX is confirmed working on INAV, but I don't see anything in INAV 6.1 to enable ADSB nor are the CLI commands listed valid. Does this mean that link was just a wish list of how it would have worked?

wx4cb, 2023-05-28:
> I am new to github in general, this link makes it seem like it was working with INAV? Just going by it saying the PingRX is confirmed working on INAV, but I don't see anything in INAV 6.1 to enable [...]

no,.. as i said before, it's not in the 6.1 firmware it's in the companion ground station software called MWP. i have no idea of who "shikofthera" is but that's obviously a forked version of inav.

you can find MWP here: https://github.com/stronnag/mwptools

dcan999, 2023-05-28:
> no,.. as i said before, it's not in the 6.1 firmware it's in the companion ground station software called MWP. i have no idea of who "shikofthera" is but that's obviously a forked version of inav. [...]

Alright thank you for the info. I don't fly with a ground station, I current just leave a ADS-B radar app open on my phone, but that isn't as effective when you are miles from launch point, which I think would be the same problem using the MWP since it's tracking aircraft proximity to the pilot, not the drone.

ShikOfTheRa, 2023-05-30:
Hi - I submitted a PR some time back for this but unfortunately it was rejected because of formatting.. 

Unfortunately I applied reverse polarity to my unit and can't afford to purchase another to do the testing.... 

It was really nice flying with it too. I never got close enough to an aircraft in the air to trigger the warning, but it was really interesting to see proximity on the display.

jeffalanmcclain, 2023-07-25:
Hum...really would like to see this supported in iNav.  DJI had ADS-B monitoring and showing the aircraft in the mini-map natively on my old Air2.  I miss it on my mini 3 Pro.  Not sure if the new Air3 will have it, but it would be very useful.  And as long as they are requiring Remote ID, might as well come up with a way to show that too!  LoL

dcan999, 2023-07-28:
> Hum...really would like to see this supported in iNav. DJI had ADS-B monitoring and showing the aircraft in the mini-map natively on my old Air2. I miss it on my mini 3 Pro. Not sure if the new Air3 [...]

When I talked to Mr.D about it he basically said it won't be supported because the PingRx is too expensive (which it is, $300 is probably more than what a lot of the models it would be installed on cost). If we could get something in the $20-50 range it would be more likely to be supported since more people would buy them, and I don't see why we couldn't. I am not an electronics expert by any means but searching around I see a few DIY ADS-B units that look like they could be mass produced should someone want to do it.

jeffalanmcclain, 2023-07-28:
I see quite a few ADS-B USB Rx's for sub $40, but I think they are primarily for use in a raspberryPi.  However, I don't think it would be much different to connect them to a UART on a FC??

error414, 2023-07-29:
I found cheaper solution TT-SC1 tybe B (with UFL connector), it's not $30 - $40, but $119, it's still much cheaper than PingRx. i have not tested it yet, I just ordered devel board. Acording manual the module should be set to mavlink mode. https://www.aerobits.pl/wp-content/uploads/ci_uploads_software/tt_sc1/technical_documentation_sc/OEM_SC_Datasheet_07o-v1.1.8.pdf It's just only theory.  Aerobits produces final receiver as well https://www.aerobits.pl/product/aero/  it costs $199, it seems to much, because it should be enought for TT-SC1 tybe B create antenna (69mm), and LDO  5V to 3.3V, it's all. 

I will see when eval board arrive.

ADSB USB dongles are much cheaper because they contain only RF Ic and ADSB decoding is in PC/Raspberry.

dcan999, 2023-07-30:
I got the feeling that unless units become 20-40$, they wont bother to support the feature. Sucks because ardupilot supports it, and there is actually a whole Cube FC with ADS-B built in for $350. I might try my hand at ardupilot since they seem to support a lot more advanced peripherals.

error414, 2023-07-30:
would be nice if some from INAV developers answered here.,

ShikOfTheRa, 2023-07-31:
I must admit as an inav guy I do actually run an aircraft with ardu solely to use ADSB for when I need to. 
Confessions over... 

I would be happy to port the previous PR to latest inav dev if someone would help with any formatting etc to get in into the main code. But I don't want to spend the time if its not going to happen.

About cost - yes, it is not cheap and its not going to be for everyone. Aerobits @ $119 is cheapest until someone DIY's a cheaper SDR based option.

error414, 2023-08-09:
I ported code from ShikOfTheRa PR to latest Inav code. It works nicely with TT-SC1 ADSB ($119) receiver. 

[image]

ShikOfTheRa, 2023-08-09:
This is great. See if I can get on a plane in next couple of weeks to show an in flight vid. Weather dependant :)

Question for iNAv devs - if PR is raised would it likely be added to inav? Subject to completing and request to improve formatting etc.

 - $119 is more realistic figure too. I would pay that for planes that cloudsurf ...
 - probably someone in the future could produce a low cost diy option using TL IC or dongle.

dcan999, 2023-08-09:
I see a lot of USB ADS-B dongles for $60-70. I admittedly don't know much about com protocols so I'm not sure how hard (if even possible) it would be to use a UART-USB bridge and one of the cheap dongles, or if the dongle itself can be converted to UART on the PCB.

ShikOfTheRa, 2023-08-09:
> I see a lot of USB ADS-B dongles for $60-70. I admittedly don't know much about com protocols so I'm not sure how hard (if even possible) it would be to use a UART-USB bridge and one of the cheap dongles, or if the dongle itself can be converted to UART on the PCB.

You would need something to act as USB host, set up the dongle SDR and process data received. Possible yes, but its a lot of effort unless someone really wants to do it. This is why those modules cost so much - they have to recover the cost of development time to develop IC / FPGA. Maybe one day community will do something, but I thing it will not be the size of the solution error414 mentioned!

$119 is pretty fair price

dcan999, 2023-08-09:
I see. I am looking at the TT-SC1 and it looks like they have the bare module/board for $119 or a complete unit (Aero) with connector for $200..what am I getting for the extra $81? A 5v-3.3v step down and JST connector? It looks like the bare SC1 module you'd just have to connect 3.3v (available on my FC), ground, and TX/RX, could solder wires directly to the board it comes on, no? Would save $80 vs the Aero.

error414, 2023-08-10:
You are right, all "magic" is in module TT-SC1, for $81 you get DCDC 3 -> 3.3V, JST, anetnna, shrinktube and maybe LED :D. I don't understand why unit is so expensive compare to module. 

It's not as obviously from image, the module is really small. solder wires will not be easy, but it's possible. Nice solution could be similar to BT dongles
[image]

ShikOfTheRa, 2023-08-21:
So test flying today and this aircraft was gradullay coming close. I could hear it, but would have had no idea how close without adsb.

By time I landed and put recorder on, just over 2km away and only 230m above ground

[image]

dcan999, 2023-08-22:
I hope this gets implemented in INAV 7! I know its a longshot but maybe something like Ardupilots "sense and avoid" as well.

jeffalanmcclain, 2023-08-28:
I am totally willing to spend $100 for an ADS-B Rx to be able to see feeds in my HUD in iNav, so I REALLY hope they support the PR on this.  I need to find a good source in the US for those Rx's.  I'm now also doing ADS-B feeding on my RPi 4 with 200mi range, so hoping to be able to have one on the plane too.  Next...Remote ID monitoring in HUD!  LoL

machatron, 2023-08-29:
This is also something I'm interested in and I super appreciate the effort you put in @ShikOfTheRa and @error414. I'm glad development is active again after 3 years and this wasn't just forgotten. 

Just seeing in OSD is nice but autopilot avoidance down the line would also be a cool goal.
I'm not afraid of diying the hardware. After all it's the first step towards Chinese manufacturers like mateksys to develop such a sensor, and it would definitely bring the cost down to hobbiest levels. The Aero hardware is commercial grade and has certifications which isnt helping cost nor needed for most of use cases.

error414, 2023-10-11:
I reated PR for ADSB support https://github.com/iNavFlight/inav/pull/9355 it uses mavlinkg ADSB message so I believe that every in future ADSB device will support this kind of message. You are right, so fat it's expensive, maybe we can ask aerobits whether it;s not posible to create lightweight ADSB version for hobbyist.

closed 2024-02-02 by DzikuVx

dcan999, 2024-08-21:
I see this is marked as completed, but what was completed exactly? I just got my TT-SC1, connected to UART 5 with MAVLINK selected for that ports telemetry, but I'm not getting anything in system messages in the OSD, or radar contacts etc...was this never implemented into the official INAV code or?

ShikOfTheRa, 2024-08-22:
I think it will be in next release. Finally :)
For now you can build your own firmware from source.

It is cool feature. I like it a lot.

dcan999, 2024-08-22:
Ah, I see. I took [this](https://github-wiki-see.page/m/stronnag/mwptools/wiki/Testing-INAV%27s-ADSB-functions) web page as meaning it was operational, but I guess "OSD" in that sense means the GCS screen, not our in goggle/monitor OSD. 

I tried to build a custom target once and it didn't go well, I guess I will have to try again. Yes, I've never had a close call, but I do wonder where other planes are when I'm out there, so this feature is huge to me too.