iNavFlight/inav #5748: Better behavior when requested vtx_power is greater than the vtx_power supported by VTX

Dope-johnny, 2020-05-17:
## Current Behavior

When you have a VTX with fewer power levels than 5 (e.g. Rush Tank Ultimate Plus) and set the power level via Smartaudio to 5 it will stay at the previous power level.

## Desired Behavior

Some options:

* Define max supported power level and don't allow to select unsupported power levels or INAV dials it back to the max supported power level.
* Prompt a warning when active power level reported by VTX doesn't match requested power level
* Automatically detect highest supported power level and set it to that

## Suggested Solution

```
CLI option
vtx_power_max = 5 // default

IF vtx_power_requested > vtx_power_max
      set vtx_power = vtx_power_max
ELSE 
      set vtx_power = vtx_power_requested
```
## Who does this impact? Who is this for?

People using multiple aircrafts with different max power levels. Especially when low power disarm is enabled.

## Additional context

I know BF addressed this with the VTX tables but I also aknowledge the reasons (complicated setup) to not implement it like that.

I think it's necessary for power levels because at close range (before liftoff) you don't recognize the difference between 25mW and 800mW. When the vtx doesn't support a frequency then you will recognize it because you don't get video on that channel.

My knowledge about the vtx control protocols is limited... If SA or Tramp is bidirectional then comparing the vtx settings reported by the VTX with the requested VTX settings might be an option... but do various VTX (especially with unofficial SA/tramp implemetation) really report the correct power level / frequency they transmit? Or just report the vtx settings they received and then do something completly different (e.g. stop transmitting when selected locked channel)... just something to consider here.

I also see a problem with automatic options or warnings with low-power-disarm: When INAV switches the power level after arming then there's a good chance the aircraft is already in the air. So no benefit compared to the behavior right now - you still need to land to adjust vtx power level. Annoying with fixed wings that are hard to launch.

Scavanger, 2020-05-20:
This affects Smartaudio 2.0 devices only.
2.1 devices report thier supported power levels and you can't switch to an unsupported power level. SA 2.1 support will come with inav 2.5. See #5713 
For the next releases I plan to add an CLI command for smartaudio advanced configuration with the ability to set max power levels for 2.0 devices.

Dope-johnny, 2020-05-20:
Okay, great! Thanks for the explanation. I'm looking forward to it.

leeph, 2020-05-27:
Is SA2.1 support included in iNav 2.5 RC2 just released?

I've just been testing it on my Vortini with TBS Unify Pro32 HV and it still seems very hit and miss, with OSD not reporting power level changes (though I can see a flash when low-power disarm / rearm kicks in to change power).

Are there new CLI parameters for SA2.1 that need to be set? @Scavanger thanks for your work on this.

closed 2020-08-09 by stale[bot]