iNavFlight/inav #5750: Add 'automatic servo trim' for fixed wing

Painless360, 2020-05-18:
At the moment the trimming of the servos can be done using the Servo Autotrim function. While this works well there is a feature in ArduPlane that I use a lot that does this automatically fro the pilot (https://ardupilot.org/plane/docs/auto-trim.html)

The automatic autotrim function monitors the value of the servo channels needed to keep the model in the level attitude and then updates these refined middle channel positions for the elevators and ailerons at a regular interval. This is very handy if you set up the model with a slightly different CG - the model trims itself automatically. 

The idea is to add this feature to iNav and have it settable as 'always on' so that iNav can perform the same kind of function. 

This would make it easier for all fixed wing pilots to have the plane auto-trimmed for level flight without having to do anything and a short flight in a flight stabilised mode would set the servos for a great manual flight too. Every flight.

Thank you for your consideration, I've looked for this suggestion in the Wiki but didn't find it so apologies in advance if you already have this on the list somewhere... Happy flying!

Jetrell, 2020-05-26:
This would be a great feature. 
@digitalentity Would the AHI / Accelerometer drift bug cause such a feature to unreliable at present?

digitalentity, 2020-05-26:
Trimming is essentially "flying stable in a certain position" and I-term does exactly that dynamically. We might want to add a very slow integrator on top, to actually "trim" servo middle points in flight all the time. I don't think AHI drift will be relevant as long as we don't trim for a "level flight" specifically.

Also IMO "level flight" is unrelated to servo endpoints and more related to FC alignment (pitch specifically). From FC perspective we might be flying level, but we might be actually climbing or sinking because of incorrect angle of attack. Trimming FC pitch alignment is somehow non-trivial, but IMO doable.

FPVZaphod, 2020-10-24:
In my very fast planes (Strix Goblin and SAB Tortuga) I fly 90% of the time in manual mode, because I love to "feel" the plane. iNav is for the home arrow, and bring the bird back home in worst case. 
I was thinking already to program my radio in a way that in manual flight mode the radio trims are active, but not in all other flight modes.  Working solution, but I would prefer to have the option to trim the plane in iNav.  Two new options in "Adjustments",  "Trim Roll" and "Trim Pitch".

myrcwing, 2020-11-01:
What the original poster mentioned would be a great feature to have for many fixed wing pilots. I second this

Avegawanderer, 2021-03-07:
I think, the decision lies on top. Without FC all the mixing, throws and endpoint settings, etc is setup in the transmitter. When using FC, this part is moved to FC, leaving transmitter only stick position measuring de-facto. The trim function also becomes a function of FC, and hardware trim buttons must be simply sent to FC to adjust servos midpoints. This may be already done with opentx. The single missing part is some logic in the in-flight adjustments that changes servo midpoints.

FPVZaphod, 2021-03-07:
Yes, agree. "Simply" add three new items to the Adjustments: "Aileron Trim, Elevator Trim and Rudder Trim". Then everybody can use / assign by themself as required. 
"Simply" from users view. Might be absolutly not simple vom Coders view, I am absolutely aware of that.

Avegawanderer, 2021-03-07:
Yes, it is not that simple for implementing. I have suggested to make global variables adjustable and use them in mixer as midpoints (see #5966). Having possibility to adjust GVs would also useful for a number of other things. Having dedicated 3 servo trim is easier but less generic.

avsaase, 2021-03-13:
Does anyone know how ArduPlane does this exactly? The [docs](https://ardupilot.org/plane/docs/auto-trim.html) say:

> By setting SERVO_AUTO_TRIM to 1, ArduPlane will constantly monitor how much it needs to correct the flying trim to maintain level AHRS pitch and roll while in the aforementioned modes. Every 10 seconds, it will store this trim into the appropriate SERVOx_TRIM values for outputs assigned to control roll and pitch (ie Aileron/Elevator, Elevons, etc.). This includes flaperon and differential spoiler configurations. After a minute or so of flight, the trims will be adjusted such that minor trim errors will be canceled out when switching back to MANUAL and ACRO modes.

This suggests the current servo position will be saved as the midpoint every 10 seconds. But then why does it take a minute for the plane to be fully trimmed out? Also, there has to be some sanity checking so that the trims are not updated during hard maneuvers. I have searched through https://github.com/ArduPilot/ardupilot/tree/master/ArduPlane but I cannot find where they implemented this.

I think I have a working solution that every second takes the current servo position and compares it to the midpoint, and then adds 5% of the difference to the midpoint: `new_midpoint = old_midpoint + (current_position - old_midpoint) * 0.05`. The midpoints are only changed if the sticks are centered and the total rotation of the airplane is below a threshold (still need to find a good value for this). 

The problem is that I need to decrease the Iterm, otherwise the updates will be noticeable in flight. My current solution is to decrease the Iterm by 5%, which I hope will exactly cancel out the change in servo midpoint, but this is difficult to test on the bench.

JuliooCesarMDM, 2021-03-23:
hi @avsaase the auto-trim servo is here https://github.com/ArduPilot/ardupilot/blob/3035f8dea3ac497488fe86a6eb98a158ef6fcb46/ArduPlane/servos.cpp#L961

avsaase, 2021-03-23:
@JuliooCesarMDM Thanks. It seems I came up with basically the same idea of slowly transferring the I-term to the servo centers. I'll add a check for high enough speed as in AP.

PR #6708 (2021-05-08): Automatic servo trim
Closes #5750 and #5719.

~~When the total rotation of the airplane is near zero and the sticks are centered, the servo midpoint is adjusted in the direction of the current servo position. The I-term is scaled down accordingly to avoid jumps in the servo position. On the bench, it doesn't perfectly cancel out but in real-world flying conditions, this should not be noticeable. (The I-term seems to slowly shrink with the model sitting still when `AUTOTRIM` is not enabled. Not sure why that is.)~~

~~The servos that are trimmed are configured using an extra `<trim>` flag in th [... cut]