iNavFlight/inav #5752: Remove compass data from failsafe RTH

rFlightFPV, 2020-05-18:
## Current Behavior
Magnetometers on multirotors are susceptible to interference and might hinder a successul RTH process in a failsafe condition. Failsafe, e.g. loss of radio link, might get triggered due to external influence which also could cause the compass to produce erroneous data.

## Desired Behavior
As in Betaflight "GPS Rescue" - but for INAV exclusively in a failsafe-RTH condition - only use GPS coordinates and barometer for RTH. The return most probably won't be as smooth but in a failsafe getting the quad back (into radio range) has highest priority imho.

## Suggested Solution
For failsafe, use Betaflight "GPS rescue" code for RTH on multirotors

## Who does this impact? Who is this for?
All INAV multirotor pilots

tgreer, 2020-05-22:
@rFlightFPV if the compass is successfully working as part of the bigger picture, why exclude it from use? I don't see what you gain from dropping it only for failsafes?

Christianmix, 2020-05-26:
very good, it would be like a fixed wing, a quad flies in any direction, I don’t see the need to have to return home with its correct front, this is in case of failures, no matter how it comes back as long as it comes back.

MrD-RC, 2020-06-10:
The problem isn’t that the compass is used, its the way that it’s used. 

I would love to enable compass in fixed wing, but it adds some bad behaviours in modes like 3d cruise, waypoints, and rth. In this case, if you have no compass, the plane will fly in a straight line towards the location. If the compass is enabled, it will fly towards the heading, but not necessarily straight. So with the compass enabled, the plane could be blown off course until its flying in to wind. 

What should happen is that all available sensors are used. Not just switch to one if its installed, the other if not. Use the data from the compass, gps, baro, and estimated wind to best navigate the craft.

zoran-grujic, 2023-05-04:
My proposal:
I would change the request into: "Implement compass-free RTH like Betaflight"

Why?
It seems like compass is not useful at all. Also, in general it is NOT reliable. It requires frequent re-calibration that most people doesn't like to loose their time into, and it is usually first sensor to give false info. Also, if pilot add/change payload must/should re-calibrate. Boring! If Betaflight has RTH without compass then INAV must do it better with no compass!

HOW it works in Betaflight?
In BetaFlight, when RTH is activated, the multirotor will go to angle mode, stop - try to hover, gain the  preset altitude, and then go in forward direction whatever it is. After some distance it will learn heading from its GPS, correct itself, and go to the home position.

HOW to improve?
All multirotors have gyros and they can use GPS to learn (when velocity is above some level) proper RTH heading. Then until the next good GPS info heading info can be maintained with gyro data only. In this way the multirotor can have sufficiently good home heading to start from and latter while moving fast to HOME it will be corrected with new GPS data. The minimum RTH speed should be set above some reasonable level.

Shall I post this as new issue/request?

closed 2024-02-02 by DzikuVx