iNavFlight/inav #5753: Autotrim only applying midpoint offset to one aileron

mattcave, 2020-05-18:
## Environment

FC: Matek F722 Wing
Firmware: iNav 2.4 (believed to still be present in 2.5 RC1)
Airframe: Finwing AlbaBird. ie conventional tail aircraft.

## Current Behaviour

I have two aileron servos on outputs S1 and S2. When I auto trim, iNav applies the midpoint offset to ONE aileron but not the other. e.g. servo midpoint for one will still be 1500us, the other will be, say, 1460us. 

## Steps to Reproduce

1. Configure mixer with two ailerons on S1 and S2
2. Arm. Enter auto trim mode and apply left stick during auto trim. 
3. Disarm.
4. Examine outputs tab. One servo, in my case S2, is set to 1460us, other servo (S1) is still 1500.
5. Observe control surfaces on plane. One servo is offset as expected, the other is still at neutral position.

## Expected behavior
After auto trim, both servos should be deflected by an equal amount in opposite directions. e.g servo 1 should be at 1460us, servo 2 should be at 1540us.

## Suggested solution(s)

Looks like a bug with new midpoint only being applied to one output. 

## Additional context

CLI dump:
https://pastebin.com/RrwALRzx

Configurator Screenshots
[image]
[image]
[image]

nikram77, 2020-05-18:
I can confirm. Same behavior on my F722 Wing (also conventional tail airplane), Autotrim only applies to S2, S1 stays at 1500us.

leeph, 2020-05-20:
I can confirm same behaviour on Matek F722-Wing using V-tail preset.

Arvycka, 2020-05-20:
Restore default servo mapping from the template.

leeph, 2020-05-20:
'the template' - what template?

If you are talking about mixer presets, then I am using the mixer preset in iNav for Mini Talon V-Tail. Stabilised roll is assigned on outputs S1 and S2 by default on F722-Wing, which is where I have them plugged in, as does Matt.

You are right, the iNav mixer *is* weird, it's like the left hand doesn't know what the right hand is doing, half the time. It's not very well implemented and needs rewriting to be more logical, IMO.

Arvycka, 2020-05-20:
I mean servo numbers on the left, not the FC pads. Load mixer and apply,
then keep those numbers as they are.

leeph, 2020-05-20:
[image]

We have not changed them. The servo numbers are exactly as per the mixer preset, both for Mini Talon V-Tail and Airplane General presets.

Stabilised roll (ailerons) are on 'Servo 2' and 'Servo 3' which correspond to FC outputs S1 and S2.

nikram77, 2020-05-23:
If specific outputs need to be on specific pad/number - what`s the point of customizable mixer??
And I wan`t to remind that it isn`t happening only on F722 Wing. I reproduced this on F405 Std and Omnibus F4 Pro since my last comment. And on all this boards with 2.2.1 issue doesn`t exist (all this boards had 2.2.1 allready and worked fine), so it`s definitely 2.4 bug.

Arvycka, 2020-05-23:
If spare channels are available, I find it much better to trim via aux
channels. Autotrim as it is is difficult to use and hardly ever works
anyway because of issues like above.

leeph, 2020-05-27:
> 
> If spare channels are available, I find it much better to trim via aux channels. Autotrim as it is is difficult to use and hardly ever works anyway because of issues like above. 

This could certainly be a workaround but this issue is for rectifying the bug in the code, which still exists as of iNav 2.5 RC2.

digitalentity, 2020-05-27:
@leeph can you please share your full configuration diff?

leeph, 2020-05-27:
Of course: https://pastebin.com/c5BVUeHx

Note that although I have a slightly non-standard config for the tail servo mixer, the aileron servos are plugged into S1 and S2 which is exactly as the Airplane V-Tail preset calls for, and this is where the issue manifests. Autotrim is applied only to the left wing aileron on S2, and not the right wing aileron on S1.

closed 2020-08-09 by stale[bot]

leeph, 2020-10-21:
Can we revisit this please? There needs to be a way to define which servos get autotrimmed if one configured the mixer servo outputs differently than the platform default.