iNavFlight/inav #5754: IBUS

Neuroplant, 2020-05-18:
## Current Behavior
Use of ibus RX and ibus telemetry on only one port does not work in 2.5.0 RC1
(no telemetry no RX)
Worked fine on Versions before.
- FC Board Matek F722-SE

t413, 2020-05-19:
Same issue here with crossfire + MATEKF411SE. 

Tried setting up this new F411-WSE with the RC1 and crossfire isn't functional. Reflashed to the stable release and it works great, same wiring (never unplugged).

Happy to help test if you want to send binaries.

tgreer, 2020-05-22:
I've just tested on 2.5.0 RC1, on a Matek F722-PX with crossfire setup. Serial port set, RX configured, but Reciever tab shows 0s. Telemetry is not being sent.

tgreer, 2020-05-22:
[image]
[image]
[image]

t413, 2020-05-22:
This one is kinof a big deal, I wish we could rename the issue to **'2.5.0-RC1: Serial receivers broken for F4 & F7 boards, IBUS & crossfire appear nonfunctional'**

I wonder how the devs are flying. Here's the [diff of new commits since RC1](https://github.com/iNavFlight/inav/compare/2.5.0-RC1...development) and there's no sign of any relevant fixes yet. Perhaps it's worth creating a new issue just to make the title more imperative?

Perhaps @stronnag would be the one to involve?

stronnag, 2020-05-22:
Flew a serial RX today just fine, so serial receivers in general are not broken.

tgreer, 2020-05-22:
Any more info? Seems like the  there’s issues if 2 different protocols are broken yet completely missed.

stronnag, 2020-05-22:
It may be that ibus and / or crossfire are not working; however sbus and dsmx are working which means that serial RX *in general* are not broken. I'll maybe find an ibus rx tomorrow and have a look at that.

stronnag, 2020-05-23:
Just tested IBUS (painfully, the RX is on a non-FC plane, finding a diode to make the cable took forever). OTOH, it appear to works, RX bar display moves as expected in the configurator.

stronnag, 2020-05-23:
It would be useful if the users with non-working configurations were to post a `diff`

tgreer, 2020-05-23:
@stronnag 
# diff

# version
# INAV/MATEKF722PX 2.5.0 May  9 2020 / 15:19:46 (272de567d)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix

# servo

# logic

# gvar

# gf

# feature
feature -TX_PROF_SEL
feature -AIRMODE
feature GPS
feature PWM_OUTPUT_ENABLE

# beeper

# map
map TAER

# serial
serial 3 2 115200 115200 0 115200
serial 4 2048 115200 115200 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout

# master
set gyro_sync = OFF
set gyro_hardware_lpf = 256HZ
set acc_hardware = MPU6000
set align_mag = CW270
set mag_hardware = HMC5883
set baro_hardware = BMP280
set pitot_hardware = NONE
set receiver_type = MSP
set serialrx_provider = CRSF
set motor_pwm_rate = 8000
set motor_pwm_protocol = DSHOT300
set throttle_idle =  7.000
set motor_poles = 12
set current_meter_scale = 649
set model_preview_type = 3
set applied_defaults = 1
set gps_provider = UBLOX7
set name = Disco

# profile
profile 1

# battery_profile
battery_profile 1

set battery_capacity = 6600

# end the command batch
batch end

tgreer, 2020-05-23:
I'm wondering if it's a configurator missmatch that's causing it. I'm using the one linked for mac, but shows only 2.4.1?

stronnag, 2020-05-23:
`set receiver_type = MSP` 
That's not going to help. I wonder also if there is a configurator bug here.

stronnag, 2020-05-23:
This being the second time in 3 days I've make that comment on a `diff`

tgreer, 2020-05-23:
@stronnag I think the issue is I'm using https://inavapp.com/ which was linked/recommended somewhere, and I've no idea where, however it's only 2.4.1 not 2.5.0 which would likely be the cause. There's no proper mac release for 2.5.0 RC2 configurator.

stronnag, 2020-05-23:
Can you please

*  `set receiver_type = SERIAL`  in the CLI
* Ensure that a UART is set os serialrx (this may be default an missing in `diff`, i.e. type field is 64), again from the CLI

Then you can probably confirm its working.

tgreer, 2020-05-23:
> Can you please
> 
>     * `set receiver_type = SERIAL`  in the CLI
> 
>     * Ensure that a UART is set as serialrx (this may be default an missing in `diff`, i.e. type field is 64), again from the CLI
> 
> 
> Then you can probably confirm its working.

Yes confirmed working.

stronnag, 2020-05-23:
Cool, I think we can close this.

closed 2020-05-23 by stronnag

tgreer, 2020-05-23:
@stronnag thanks. I'm not entirely sure what page linked the inavapp application but it was linked from somewhere inav related. I didn't magically guess the URL to download it!

stronnag, 2020-05-23:
afaik the inavapp app (we love RAS) ran the configurator in the background, so I'm not sure where the fault might lie.

stronnag, 2020-05-23:
I suspect this is an unavoidable side effect of https://github.com/iNavFlight/inav-configurator/pull/933 and #5369, especially if full erase was not used. I'll add a note to RN.

stronnag, 2020-05-23:
btw there's an unsigned, unofficial, unsupported, untested, un-anything MacOS build of the latest configurator at http://seyrsnys.myzen.co.uk/inav-configurator-linux/ (sic)