iNavFlight/inav #5756: Kakute F4V2 telemetry not working

albebert, 2020-05-18:
## Current Behavior
Inav 2.4.0 Telemetry working from rx to inav using smartport (r9m) (rssi is ok on inav)
but no return to TX (Taranis x9d + r9 2019 flex)

## Steps to Reproduce

1. use a Kakute F4V2 with a frsky r9m rx
2. configure telemetry
3. see RSSI going to 0% when shutdown the TX
4. discover telemtry on tx return no value from inav

## Expected behavior
telemetry report on tx

## Additional context

- FC Board name and vendor: Holybro Kakute F4V2
- INAV version string: 2.4.0

DzikuVx, 2020-05-19:
I have Kakute F4 V2 configuraed with S.Port up and running. In your case I expect you used either a TUART without a inverter or not configured smartport invertion and half duplex correctly.
Please refer to Fc manual

closed 2020-05-19 by DzikuVx

albebert, 2020-05-22:
i use dedicated SP uart with interver (uart 1) , all is working on another firmware (emuflight for example)

telemetry inversion is off (and tested at on but no luk) and i don't see option tu set half duplex on telemetry.

i use a r9mm rx sbus on uart 3 and smartport on uart 1 (see the pinout below) both have a inverter

[image]

ch3ll0v3k, 2021-02-03:
I have the same issue.