iNavFlight/inav #5764: MATEKF765 2.5.0_RC1 no Input from SBUS Tx

Kraligor, 2020-05-20:
## Current Behavior
No input from transmitter is registered in the configurator.

## Steps to Reproduce

1. Flash MATEKF765 2.5.0 RC1
2. Bind transmitter
3. Open configurator, go to Receiver tab
4. Move controls on transmitter

## Expected behavior
Input from transmitter should be registered and shown in the Receiver tab. It works with 2.4.0.

---

No access to my FC right now, can add later if needed.

- FC Board name and vendor: MATEKSYS F765
- INAV version string: 2.5.0 RC1

digitalentity, 2020-05-20:
Please provide dump of the configuration.

Kraligor, 2020-05-20:
> Please provide dump of the configuration.

Attached to the OP.

stronnag, 2020-05-20:
Configuration error from your dump. I seriously doubt this worked in 2.4.
```
set receiver_type = MSP
```
Please reopen if this is not the case, also consider social media for future support issues.

closed 2020-05-20 by stronnag