iNavFlight/inav #5767: Reversible ESCs not working correctly on rover

WizzX-FPV, 2020-05-22:
## Current Behavior
Im using INAV on my experimental rover/crawler (FY03) and I can see that there's issue with controlling motor reversible (works fine in forward only)

Seems like 0 point of throttle is quite out of center no matter which walue I set as neutral position or low/high deadband in INAV. also I can not regulate speed of the motor starting at minimal rpm - only starts in som specific (quite high) speed around 30%

## Steps to Reproduce
If I arm the rover with throttle stick in middle position than motors will start to turns backward in a quite high speed.
pushing a stick more up (higher throttle) will hardly break the motor to stop and there's a dead band at around 1550 to 1650 area. after position of 1650 motor will start to turn forward but starts quite fast.

I had no issue with older version of INAV 2.4 but I did not make a backup of that before update to 2.5 RC1

## Expected behavior
stop the motors at 1500 with small deadband and softly start forward/backward from mimimal possible speed of motor

- FC Board name and vendor:
MATK F722 std
- INAV version string:
INAV 2.5.0-RC1

- Im using bidirectional connection of ESC 4khz and dshot 600 ESC
- ESC: DYS Aria BLHeli_32bit 70A ESC
Rampup power 70%
Sine mode:ON
Direction: Bidirectional Soft
Timing: 30 deg
Stall: relaxed
Brake on stop: 20%

- Motor: 3650 2300KV (4magnets)

WizzX-FPV, 2020-05-23:
I will try to describe more details and add my diff file.
Seems that is somehow related to idle % I set 4% but there's still issue and most important offset from stick position to motor output.

I made a table comparison on reciever stick position vs motor output
receiver | motor output
1000 | 1034
1100 | 1034
1500 | 1396
1560 | 1450
1561 | 1490
1670 | 1490
1671 | 1550
2000 | 1850

- You can see that there's deadband from stick position 1000 to 1100 (backward direction)

- Than in middle position of stick 1500 motor is still turning backwards (motor output 1396).

- Motor (hardly) stops turning if stick position reach 1561 and there's deadband to stick position of 1670 (motor output is 1490 if stick goes from down to up and 1510 if stick goes from up to down - seems like acording 3D deadband settings) 

- Than motor starts to move forward when stick is in 1671

- On max speed there's only 1850 motor output even if stick is in 2000

As i would like to use my rover as crawler, I need to run motors as slow as possible

my diff
[INAV_cli_KRAVA_FY3_20200523_095745.txt](https://github.com/iNavFlight/inav/files/4671385/INAV_cli_KRAVA_FY3_20200523_095745.txt)

wx4cb, 2020-05-24:
speaking of which. if im using a standard rc car ESC (brushed - need to replace the esc to a decent one) is there a way to have 0 - 1500 to be reverse ? I know pawel has to re do the esc stuff i beleive unless someone knows how to do it for a brushed rc car esc ?

DzikuVx, 2020-05-25:
Reversible ESCs in 2.4 were not working correctly. 2.5 is closer to correct behavior but we probably still miss the option to reverse in firmware.

Now, for the neutral and throttle mapping.
1. Throttle is mapped directly to ESC output and it's not 1:1
2. there are 3 settings that describe neutral point of ESC: `3d_neutral`, `3d_deadband_low` and `3d_deadband_high`
3. They have to be adjusted to have motors spinning correctly:
4. Adjust `3d_neutral` so that after arming motors do not spin
5. adjust `3d_deadband_low` and `3d_deadband_high`so that after you move throttle they spin with desired minimal speed

I will try to add option to reverse reversible ESCs soon

WizzX-FPV, 2020-05-29:
Thanks Pawel,
Its partly works fine for me now.
It was just my different understanding of "3d_deadband_low and 3d_deadband_high" as there's probably some hard coded dead bands which i needs to fits with those two parameters.

I did workaround for keep correct offset that in the middle position of stick motor stops:
* set Throttle MID to 1 and Throttle EXPO to 0.34 in Receiver tab.
Then sticks curve is different for forward/backward, but it works.

There's still quite big dead band (1450-1550) but I can start to move motor from min speed in both direction.
what I still can see as unclear is:
* why max throttle is only 1850 and min 1034?
* why "Motor IDLE power [%]" can not be set below 4%? for rover i would like to disable that (go to 0%).

DzikuVx, 2020-05-29:
@WizzX-FPV for INAV 2.5:

1.  "Motor IDLE power [%]" is not used for Reversibe ESCs. 
2. Output is scaled between `3d_deadband_high` and `max_throttle` for FORWARD direction
3. Output is scaled between `min_command` and `3d_deadband_low` for BACKWARD
4. When motor is supposed to STOP, `3d_neutral` is sent to ESC
5. When there is no MOTOR_STOP output never goes into the `3d_deadband_low` : `3d_deadband_high` range. When motor should reverse direction, it just switches to the other range. 
6. THROTTLE input and ESC output are separate entities with separate ranges. THROTTLE range is scaled to OUTPUT range

closed 2020-06-09 by DzikuVx