iNavFlight/inav #5770: 2.5.0-RC1 bug with "feature THR_VBAT_COMP"

dsaboot, 2020-05-23:
## Current Behavior

If I enable feature THR_VBAT_COMP - multicopter can't reach normal RPM of motors, can't fly, it shows up to 100% values for motors, but rotate very slowly (not enought to start to fly)
It's the same behavior for any mode ANGLE, ACRO
In 2.4.0 with same configutation was ok.

## Steps to Reproduce

1. enable "feature THR_VBAT_COMP"
2.
3.
4.

## Expected behavior

I expected the same behavior as in 2.4.0

## Suggested solution(s)

## Additional context

---

# version
# INAV/MATEKF722SE 2.5.0 May  9 2020 / 15:10:58 (272de567d)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

- FC Board name and vendor: Matek F722SE
- INAV version string: 2.5.0-RC1

b14ckyy, 2020-05-23:
Can Confirm issue on RC1 in plane config on F411WSE. 
Throttle can be controlled in outputs tab. but when armed and throttle applied with transmitter, it just goes up to idle throttle but not further. if VBAT Comp is disabled, it works as expected.

DzikuVx, 2020-05-24:
This was solved by #5726 and will be fixed in RC2

closed 2020-05-24 by DzikuVx