iNavFlight/inav #5772: Problem with INAV 2.5 ALTHOLD and POSHOLD

Igorx64, 2020-05-24:
Hello, im new to INAV, recently i build an Martian 2 drone with Omnibus f4 v3 FC and BN-880 GPS module, im having problems with navigation modes on inav, i triend INAV 2.4 and 2.5 RC1 and it hapens on both, first Altitude data is wrong if GPS vertical position weight is diferen than  0, if Vertical position barometer weight  is 1 and gps is 0, the altitude data seems reasonable, so i think Baro is working good and i put some foam shielding on it, but if GPS vertical position weight and Vertical position barometer weight ar at default values, when flying in acro the Altitude data is all wrong like -14m, 23m, -4m. Also when i manage to use only baro  to get Altitude data, the Althold and Poshold modes doesnt work as intendet, the quad start to trhorttle  up and rush to the skies, moving the throttle stick dont do nothing, hower throttle is set to around 30%,
tried with Use mid. throttle for ALTHOLD ON and OFF, triend changing weight values in Position Estimator, tried to switch Use GPS data for velocity calculation, still dont work, i  dont know what is wrong, i didnt cahnge Althold related PIDs,  can some one help my please, i dont know what elese i can do ?

Thanks.
- FC Board name and vendor: Omnibus f4 v3
- INAV version string: 2.5 rc1

brainbubblersbest, 2020-06-05:
Same issue has a User at Facebook with Standard pids and good looking parameters. Matek f722-px

matthewrwright, 2020-06-08:
Got a +1 here on this - also weirdly when I analyse the blackbox log I'm seeing a motor missing! I know for sure that motor is running as I'm watching it spin up faster and it takes off without pitching over anywhere.

Running 2.5RC2 on a Matek F722-STD - have updated to 2.5 Release but haven't had chance to fly it yet.
I was upgrading the quad from an F1 board running 1.7.3 to an F7 board with 2.5, so assumed I'd broken something in the upgrade, but I can swap the wires straight back to the F1 board and the problem goes away so I'm reasonably sure the hardware is good.