iNavFlight/inav #5774: navTgtPos suddenly goes from 5 meters to 23 meters

ulltieri, 2020-05-24:
Hello!
Today when I flew with my drone drone (Tyro129) in WP mode it suddenly rose, I changed to NAVPOS HOLD but it continued to rise. Was forced to disarm and it landed in some bushes with two broken propellers :-)
Had flown WP mode and NAVPOS hold without problems at about 5 meters altitude. But when I flew at higher altitude with NAVPOS hold and changed to WP mode and the drone was heading down to 5 meters the problem arose.
This has happened allso on another occasion under similar circumstances, NAVPOS hold to WP mode from a higher altitude to a lower.
When I look in the log I see that navTgtPos suddenly goes from 5 meters to 23 meters, at the time 4: 46.1. But I don't understand everything in the log so I would be grateful if anyone could help me with this problem. Br. Erik
[LOG00013.zip](https://github.com/iNavFlight/inav/files/4674223/LOG00013.zip)

ulltieri, 2020-05-25:
Update..
I made a CSV export and looked in Excel and see that navTgtPos[0], navTgtPos[1], navTgtPos[2] makes a big jump in just one update. Can't see any other values that make a "jump" at the same time. Looks strange, a setpoint / target value should not be allowed to make such big "jumps" without changing flight mode.
Br. Erik

[image]

digitalentity, 2020-05-26:
What mission you were flying? Specifically what type of waypoint the drone was flying when this happened?

This looks like a genuine mission progression and the drone was merely flying to the next waypoint. Unfortunately we don't log waypoint change events (something we should be doing, actually).

stronnag, 2020-05-26:
In general, it also be useful to have the XML mission file when investigating WP related reports.

ulltieri, 2020-05-26:
I was flying a short WP mission att 5m and 2m/s, just 4 WP. I did fly the same mission several times with no problem, its in the log file. But this time i started (change from nav pos hold to WP mission) the mission at a higher altitude. Can be that the drone was changing WP but it was on its way down to 5 meters.  

[mission_13.zip](https://github.com/iNavFlight/inav/files/4681066/mission_13.zip)

closed 2020-08-08 by stale[bot]