iNavFlight/inav #5778: Auto Telemetry improvement: separating ESC telemetry feedback from each other

vonkoda, 2020-05-26:
**## Current Behavior**
All Auto Telemetry ESC telemetry is just dumped together and output from the ESC/Flightcontroller. With a 4-1 ESC, its impossible to physically separate the telemetry lines. 

**## Desired Behavior**
Ability to know the Telemetry per ESC. 

**## Suggested Solution**
Ideally: Relevant identifier per ESC (Eg either the name of the ESC as its flashed or ESC1/2/3/4)
Minimum: Any ability to uniquely ID the feedback so the user can figure out which Telemetry is for which ESC. Obviously this would need to persist between sessions. 

**## Who does this impact? Who is this for?**
My use case is using the ESC's in a non-flight application (4x ESC's to Raspberry Pi) so I cannot/do not use a flight controller. Auto Telemetry allows me to avoid having to build physical temp and (e)RPM sensors. 
There are no doubt other more normal flight related use cases too, as-is Auto Telemetry that combines all feedback of 4 discrete motors/ESCs doesnt seem overly useful.

giacomo892, 2020-05-26:
AFAIK the ESC doesn't send any identifier so we are unable to tell from which motor the info comes.
A solution can be to use one UART for each motor which is a waste of resources.

digitalentity, 2020-05-26:
Unless we use polled telemetry (DSHOT-triggered) or use separate wires to receive telemetry for each ESC individually this is impossible. ESCs don't send any identifier inside their telemetry packet.

closed 2020-05-26 by digitalentity