iNavFlight/inav #5779: Inav 2.5 blinking OSD

donchia, 2020-05-27:
## Current Behavior

Matek F405 STD blinking OSD
There was no such issue in inav 2.4 and previous

## Steps to Reproduce

1.  Set PAL, NTSC, or Auto in inav config, does not matter. Camera is set to PAL.
2. Fly normally your quad in (without freestyle, or acro  )
3.OSD starts to blink sometimes
4.

## Expected behavior

OSD should not blink

## Suggested solution(s)

## Additional context

https://pastebin.com/BG1rhs4U
---

# INAV/MATEKF405 2.5.0 May  9 2020 / 15:05:07 (272de567d)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

- FC Board name and vendor: Matek F405 STD
- INAV version string: 272de567d

kibinek, 2020-05-28:
Hey, I have the same issue in BetaFlight 4.1.5. After I hit the tree... Have you checked your OSD chip physically? Could be it's damaged (ripped pin or whatever).

DzikuVx, 2020-05-28:
OSD blinnking is caused by:
- electrical noise
- ground problems
- wiring problems
- hardware failures

closed 2020-05-28 by DzikuVx