iNavFlight/inav #5780: Phantom channel inputs and telemetry issue with R9 mini reciever

donchia, 2020-05-27:
## Current Behavior

I am getting random inputs to FC from R9 mini reciever. Quad starts to twitch on one of the channels - on yaw, or pitch, or roll.  At the same time, Horus X10 start to complain: telemetry lost, recovered every second. Telemetry connected via softserial, disconnection of the cable did not helped. Quad becomes almost uncontrollable. Balckbox shows RCchannel inputs. Reciever connected via USB, cable twisted.

Using R9 mini, Matek F405 STD on quadcopter. 

## Steps to Reproduce

1. Arm and fly quad
2.
3.
4.

## Expected behavior

No random inputs, and no telemetry issues

## Suggested solution(s)

## Additional context

https://pastebin.com/BG1rhs4U
---

# version
# INAV/MATEKF405 2.5.0 May  9 2020 / 15:05:07 (272de567d)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
- FC Board name and vendor: Matek F405 STD
- INAV version string: INAV/MATEKF405 2.5.0 May  9 2020 / 15:05:07 (272de567d)

giacomo892, 2020-05-27:
Try to change sdcard and reduce logging to max 1KHz.

donchia, 2020-05-27:
Thanks for idea, 
Hm, my issue is related with copter  random twitches and telemetry problems.
Is it related to logging, how?

leeph, 2020-05-27:
This is more likely to be an issue specific to your R9 receiver / module combination.

R9Mini/MM in certain batches are known to have problems. Replace with a known-good receiver and see if problem persists. I don't see anything here that suggests an iNav codebase issue.

DzikuVx, 2020-06-11:
This is TX/RX issue or potentially caused by SoftwareSerial usage. Tested with R9Mini and R9MM

closed 2020-06-11 by DzikuVx