iNavFlight/inav #5795: OSD alignment and glitchs with Matek F722-PX

Unknown601, 2020-05-31:
## Current Behavior
OSD alignment is off and pixel OSD glitchs

screenshots https://imgur.com/a/sKZZRPK
DVR https://youtu.be/sIRV0clV3i8

## Steps to Reproduce
1. Install inav on Matek F722-PX
2.
3.
4.

## Expected behavior
OSD elements aligned with preview in the configurator.

Quadcopter Specs:
Matek F722-PX
TBS Unify Pro32 HV
Caddex Baby Rattle

## Suggested solution(s)
Ability to align the OSD in the cli.

## Additional context
https://pastebin.com/Nk8NibFS

# INAV/MATEKF722PX 2.4.0 Feb 11 2020 / 23:11:32 (bcd4caef9)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

ShawnzAZ, 2020-06-06:
I came here to report the same.  this feels similar to the DJI OSD issues where it freezes / doesn't update, but the Pixel OSD Flavor.

The big thing I can re-produce is upon disarming the craft disarms but the OSD still looks like it is armed, and no status screen.  Other times if I get the status screen it frozen on the screen and will not go away until a power cycle.

Generally with the OSD I move elements in the configurator to the far left and there is at least 1 character's worth of space along the left edge of the NTSC frame not being used.   I've also found that you cannot use the bottom line of the OSD frame in the configurator.   It just will not appear on the screen at all.

I'm also using Matelk F722-PX.  I will edit up and post some video demonstrating the issue.

I've seen the behavior on both 2.4.0 and 2.5 RC2.

Seba-VT, 2020-06-17:
I have the same issue. In ma case with the F765-Wing + ANALOG PIXEL OSD MODULE. I made a video describing the issue.

[https://www.youtube.com/watch?v=zhsbP_MNjMg](url)

closed 2020-08-31 by stale[bot]

dilberry, 2020-09-28:
I had some success by updating the Pixel OSD firmware (using the configurator from here https://github.com/FrSkyRC/FrSkyOSDApp/releases) and setting the signal type in the camera (Runcam Phoenix 2) to PAL instead of NTSC.