iNavFlight/inav #5799: Add electronic reporting according to new europeen law

activa73, 2020-05-31:
## Current Behavior

No current behavior
## Desired Behavior

The goal is to implement a protocole to dirve an external wifi module like ESP32 for exemple to send data according to the futur european law for drone > 800 g.In french this law start 29 June. This trame must have gps to send speed, longitude, latitude, height,... to the wifi module.
Input variable must be the ID of the plane.
## Suggested Solution

This web site explain how to make an  electronic reporting module with open source ressources.
https://www.helicomicro.com/2020/05/26/une-solution-open-source-pour-le-signalement-electronique-a-distance-a-moins-de-40-e/
Code to communicate with wifi module is implemented in GPL licence.
Here the code : https://github.com/khancyr/droneID_FR

## Who does this impact? Who is this for?

For all European user
## Additional context

digitalentity, 2020-06-01:
INAV is already able to output telemetry, whichc contains all information required to implement this kind of beacon. Interpreting the telemetry and further transmitting it via WiFi or BLE is up entirely up to the beacon firmware and out of scope for INAV.

activa73, 2020-06-01:
Yes I know t is not possible to controle the firmware of beacon chip directly by inav. I just want to resend data come from inav gps to esp32 chip. The goal is to connect fc board to esp32 using serial port (RX TX) to resend gps data to esp32. After we can develop firmware on esp32 to read gps data and format trame according to law.
What do you think?

digitalentity, 2020-06-01:
Reuse one of the existing telemetries - MAVLink, LTM or other.

activa73, 2020-06-01:
Ok thank you, I will search in this direction. With your experience, What is the best telemetry for my situation?

wx4cb, 2020-06-02:
mavlink would probably be your best bet as it's what ardupiot and others use

activa73, 2020-06-03:
Ok thank you.

closed 2020-08-16 by stale[bot]