iNavFlight/inav #5800: Rudder counterating turn and overreacting in navigation modes

avs2, 2020-05-31:
## Current Behavior

A while ago I noticed that on fixed wing, when I flicked into one of the navigation modes, the aircraft quickly turned left and right, and eventually spiralled out of control. It seemed that the rudder was overreacting, but since my board doesn't have a blackbox I could not confirm this. 

I posted in the rcgroups thread (starting [here](https://www.rcgroups.com/forums/showpost.php?p=44516907&postcount=25829)) about this problem but was unable to figure out the cause of the problem and kind of gave up on it. I then found a video by Andrew Newton ([link](https://youtu.be/pcgsFeRINFo?t=162)) that seemed to show the same behaviour, so I commented on it. Today he posted a video ([link](https://www.youtube.com/watch?v=RzWn-VABREc)) that shows that the rudder is counteracting turns in ACRO mode, and the overreacting of the rudder on the navigation modes. Turning the P and I gains to zero fixed the counteracting the turns (at the cost of losing the yaw stabilization) and reducing the FF solves the overreacting of rudder in nav modes (at the cost of having only small throws in ACRO mode). 

I am not sure if this is one bug or two separate bugs that both involve rudder. 

Setting `nav_fw_control_smoothness` to a higher than zero value does not solve the issue.

## Steps to Reproduce

1. Fly fixed wing with default PIFF's
2. Make aileron turns
3. Observe that rudder tries to counteract the turns
4. Flick into RTH of Pos Hold
5. Notice that the aircraft becomes unstable.
6. Disable navigation mode and see that the plane recovers and avoid a crash.

## Expected behavior

In aileron turns, ACRO does not try to counteract the turn by applying opposite rudder.
In navigation modes, the rudder is not used at all (at least, I thought this is a new feature in 2.5, so should not happen in 2.4). At the very least, rudder only helps in the turns.

## Suggested solution(s)

With my limited knowledge of control theory, I can see that ACRO will try to stabilize the uncommanded yaw from an aileron turn, but I think INAV should somehow know that you are trying to turn and not work against it.

For the navigation modes, maybe a fixed but configurable rudder/aileron mix could be used in the turns.

## Additional context

[Link to latest diff](https://pastebin.com/eb8MgYNy). This board does not have a blackbox, so no log unfortunately.

---

- FC Board name and vendor: Matek F411-WSE
- INAV version string: NAV/MATEKF411SE 2.4.0 Feb 11 2020 / 23:07:15 (bcd4caef9)

digitalentity, 2020-06-01:
This is WAI. Acro is controlling 3 _independent_ axis (think aerobatic flight, where you actually don't want the plane to change heading when banking).

Also, INAV is already able to do a coordinated turn with `TURN ASSIST` mode. Navigation modes make use of it automatically.

closed 2020-06-01 by digitalentity

avs2, 2020-06-01:
Did you look at the video by Andrew Newton? This cannot be desired behaviour for an acro mode. It basically makes it impossible to fly bank and jank. I'm not asking for coordinated turns in acro mode. I suppose I can live with zero yaw I gain as you suggested in #1297. 

Further, what about the rudder kicking the aircraft into a spiral in navigation modes, also in his video? This makes the navigation modes very unreliable, which defeats their purpose. He didn't show it, but the same problem occurs in cruise mode. See also my dvr video [here](https://www.youtube.com/watch?v=KX7wfJpwbRE) and [here](https://www.youtube.com/watch?v=309r1prSiu8).

digitalentity, 2020-06-01:
An aerobatic plane is supposed to react independently on each axis, otherwise it will be impossible to fly "knife edge" as the plane will instead enter a banked turn.

ACRO is aerobatic flight mode, hence it will maintain the aircraft orientation in space. If you want to turn you need to coordinate rudder,  ailerons and elevator. This is the way how ACRO mode is designed to work. If you want to have no stabilization on rudder - zero P and I and leave only FF the gain, but again, this will defeat the very  idea of ACRO mode.

As for the navigation instability - blackbox log will be required to debug this, but it seems that your yaw FF gain is too high and the plane actually overreacts in automated modes. Normally P and I will compensate for overreaction.

avs2, 2020-06-01:
I use ACRO mode as a sort of stabilized but not auto leveling mode, as in, I want to to counteract wind gusts etc. but otherwise have controls similar to manual mode. I think most people you ACRO like this. I don't think `TURN ASSIST` is the solution here, but the documentation is not super clear.

If you look at my `diff all` and the piffs on the osd in the second dvr video I linked, you will see that I am using the default piff's. Decreasing the FF indeed seems to solve the overreacting issue in Andrew's video, but then you lose a lot of throws in acro mode. My understanding from one of Pawel's video's is that FF just determines the throws in the stabilized modes, is this not correct?

digitalentity, 2020-06-01:
>  My understanding from one of Pawel's video's is that FF just determines the throws in the stabilized modes, is this not correct?

Not alone.  Throw is defined by `FF * targetRate[deg/s]`, make sure max rates match capabilities of the aircraft. TL;DR - max rate should be actual rate of rotation at maximum servo throw.

> EDIT: one additional point is that in the navigation modes the bank angle violates the constraint set by nav_fw_bank_angle. I think that should not be allowed to happen.

Blackbox required to confirm and further debug this.

avs2, 2020-06-01:
For the rudder action in ACRO, if this is actually desirable for acro pilots, then I think there should be  a STABILIZE mode for my use case. Tweaking the piffs to get behavior that a majority of pilots want is not very user friendly.

For the navigation instability, is it not possible to implement a simple rudder-to-aileron mix like some people do in their transmitter if they are not using a flight controller? Of course this mix should only be used for turns, not levelling the plane.

> Blackbox required to confirm and further debug this.

This board doesn't have a blackbox.

digitalentity, 2020-06-01:
> For the navigation instability, is it not possible to implement a simple rudder-to-aileron mix like some people do in their transmitter if they are not using a flight controller? Of course this mix should only be used for turns, not levelling the plane.

Navigation does coordinated turns by default. Rudder, elevator and ailerons work together to make a banked turn. No mixing required.

avs2, 2020-06-01:
But for some reason it applies way to much rudder In there turns. Are there any parameters that control this?

digitalentity, 2020-06-01:
Way too much rudder is usually caused by:

1. Too much FF gain
2. Incorrect rate settings (not matching the capabilities of the aircraft)
3. Incorrect cruise airspeed setting (`fw_reference_airspeed` parameter, should roughly match the airspeed at `nav_fw_cruise_thr`)

Without the blackbox it's impossible to tell exactly what's happening.

avs2, 2020-06-01:
I will experiment with this and report back. I damaged my only 4S lipo last time I flew so it might take a week or two.

Getting the actual rate setting can be a bit cumbersome, especially on yaw, because you need to counteract the roll. When guestimating, is is better to slightly over or underestimate the rates?

Is `fw_reference_airspeed` in cm/s, so that the default is 10m/s? The documentation does not explicitly mention this. If this is a crucial setting for the navigation modes to work, it might be good to have the parameter be configurable via the gui in the configurator. 

Finally, maybe you have already seen this, but in the rcgroups thread starting [here](https://www.rcgroups.com/forums/showthread.php?2495732-Cleanflight-iNav-%28navigation-rewrite%29-project/page1728#post44660503) I posted about this issue as well and got a good reply from notamedclosed about ACRO behavior that maybe explains my problem better than I did.

digitalentity, 2020-06-01:
> Getting the actual rate setting can be a bit cumbersome, especially on yaw, because you need to counteract the roll. When guestimating, is is better to slightly over or underestimate the rates?

When guesstimating it's better to set a higher value for max rate. Tuning FF so the resulting rate matches command is more important though.

> Is fw_reference_airspeed in cm/s, so that the default is 10m/s? The documentation does not explicitly mention this. If this is a crucial setting for the navigation modes to work, it might be good to have the parameter be configurable via the gui in the configurator.

Yes, exactly, it's [cm/s]

> Finally, maybe you have already seen this, but in the rcgroups thread starting here I posted about this issue as well and got a good reply from notamedclosed about ACRO behavior that maybe explains my problem better than I did.

The fundamental problem is that when you bank, a flight controller has no idea whether you want to fly straight on a "knife edge" or do a banked turn.

`ACRO` alone will behave the way a 3D pilot will expect a plane to behave - every stick is controlling exactly one axis of the aircraft, no turning when banking, etc.

For the behavior you desire (doing a banked turn when rolling) we have `TURN ASSIST` mode.  That mode, when enabled, will guesstimate the expected turn rate based on `fw_reference_airspeed` and actual aircraft bank angle, and enforce that turn rate by adding a required amount of rudder and elevator without the pilot touching the yaw control.

avs2, 2020-06-01:
I am now experimenting on the bench with TURN ASSIST. Is it correct that the expected turn rate is independent of the rate settings? Holding the plane in my hand at a 45 degree bank and then turning it in a yaw motion as it would in a turn, centers the rudder at equal yaw rates, I think. I also wonder how TURN ASSIST calculates the expected turn rate based on a reference air speed, even more when you have no airspeeds sensor installed. 

I hope I can get TURN ASSIST to work nicely because that indeed seems close to what is want.

digitalentity, 2020-06-02:
> Is it correct that the expected turn rate is independent of the rate settings?

TURN ASSIST calculates the turn rate independently of rate setting, but the rate settings are still applied as constraints.

> I also wonder how TURN ASSIST calculates the expected turn rate based on only a reference air speed, even more when you have no airspeeds sensor installed.

If there is no airspeed sensor, TURN ASSIST will **assume** that the plane is flying exactly at `fw_reference_airspeed`. If you're flying slower or faster - the calculated turn rate will be incorrect, that's why the recommendation is to set `fw_reference_airspeed` to a value that corresponds to `nav_fw_cruise_thr` so in cruise or other navigation modes it will be as close to reality as possible.

avs2, 2020-06-02:
I see, thanks. Looking into `/src/main/flight/pid.c` to better understand `TURN ASSIST` and I think this looks promising. Is it correct that you apply `TURN ASSIST` on top of ACRO/AIR or ANGLE, and not as a self contained mode?

I need to experiment with this mode, but if it works if I understand it now I think this maybe should be enabled by default on fw. Or at least it should get more attention in the wiki. `fw_reference_airspeed` would then need a place in the configurator, close the the `nav_fw_cruise_thr` setting. I am willing to rewrite some of the wiki to make it easier to understand to the beginner.

Small note, I notice that [line 644](https://github.com/iNavFlight/inav/blob/b7516be4a585685ee061bcc711a73f1e54c30d16/src/main/flight/pid.c#L644) says
```
float airspeedForCoordinatedTurn = 1200.0f; // FIXME, default to 43km/h at the moment
```
so the default is 1200, not 1000? Or is the default reset somewhere else?

Thanks a lot for all the explanation here and in the rcgroups thread. I learned a lot about how flight controllers work and have some things to tweak to make my airplanes fly better.