iNavFlight/inav #5801: Requesting JHEF7DUAL flight controller target to be added

JustinBluntFPV, 2020-06-01:
I would like to request the JHEF7DUAL flight controller to be added to the targets list please.  Or can someone please tell me where I can go to request this firmware to be compiled.  thanx

digitalentity, 2020-06-01:
Please read the https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md and reach out to coredev@inavflight.com to provide specs and arrange FC shipment.

closed 2020-08-15 by stale[bot]

dmak, 2025-05-16:
I wonder why specs cannot be taken from [Betaflight sources](https://betaflight.com/docs/wiki/boards/archive/JHEF7DUAL), where this board is supported?

sensei-hacker, 2025-05-16:
> I wonder why specs cannot be taken from [Betaflight sources](https://betaflight.com/docs/wiki/boards/archive/JHEF7DUAL), which supports this board?

A few reasons. We do extensive testing to make sure every pad on the board works correctly, as you can see here for this JHEMCU board I did recently:

https://github.com/iNavFlight/inav/pull/10825

Why do we need to test anything, rather than blindly copying whatever is in Betaflight? First, because INAV is not Betaflight. Things are not always the same. Indeed it's not at all uncommon to have differences. For example, as I recall that JHEMCU F405 board has two UARTs under Betaflight. It has SEVEN UARTs under INAV.

Two is probably fine for most Betaflight users, who only need to connect a receiver. INAV users will also need to connect at least a GPS, and will likely want to connect a lot more. Two UARTs isn't nearly enough for what INAV can do.

Secondly, even where things should be the same, Betaflight is not necessarily correct. It's not all that unusual for us to find problems that *also* affect Betaflight, like pins being defined backwards. So that's why we do all of the testing and make an INAV target, which properly supports the full functionality of INAV.