iNavFlight/inav #5802: 2.5-RC2 Reversible Throttle Not Going to Full Power

KenImhof, 2020-06-02:
## Current Behavior

When throttle is reversed, throttle only goes to 1150ms and not to full reverse power of 1000ms.

## Steps to Reproduce

1. Erase and flash 2.5-RC2
2. Enable Reversible Motor Mode
3. Set max_throttle = 2000 to get fully positive throttle

## Expected behavior

From the motor output window, motors should go to 2000ms with full positive throttle and 1000ms at full reverse throttle, but reverse throttle only goes to 1150ms.

## Suggested solution(s)

Problem is the generateThrottleCurve is using a minThrottle of around 1150 with the added idle percentage.  One solution is to use motorConfig()->mincommand as the min_throttle has been removed from this version.

KenImhof, 2020-06-03:
This change will also address #1879 .

DzikuVx, 2020-08-09:
With #6030 `throttle_idle` can now be set to `0` what allows full output from `min_command` to `max_throttle`

PR #6030 (2020-08-10): Allow throttle_idle to be 0 for better 3D compatibility
fixes #5802

In reversible mode, `throttle_idle` can now be set to `0` what allows full output from `min_command` to `max_throttle`
closed 2020-08-09 by DzikuVx