iNavFlight/inav #5803: Strange Magnetometer/compass behavior with F722-PX.

Unknown601, 2020-06-02:
## Current Behavior
Strange Magnetometer/compass behavior with F722-PX.
I can not get any compass to work with my F722-px.

As the quadcopter rotates the difference between its real rotations and the recorded magnetic position get more and more mismatched, It also seems to drift around sometimes.

1. I have tried every angle. 
2. Calibrated about 30 times inside and outside.
3. Tried reflashing the board with 2.4 and 2.5 -  4 times. {Full chip erase}
4. Tried a Matek M8Q-5883 and a BN880 - same problem.
5. Moved GPS module away from anything metal on the quadcopter.

This is what is happening:
[blob:https://imgur.com/dc643ddf-f2b2-47fe-94a2-d02323f2e709](url)
[blob:https://imgur.com/f8e4eb80-6b88-4714-bce2-18374b24f377](url)

## Additional context
There are 45 IC2 errors when connected to the configurator, is this important?
CLI DUMP https://pastebin.com/iFXEpevy
Wiring: [http://www.mateksys.com/wp-content/uploads/2020/01/F722-PX_wiring2.jpg](url)

---
# version
# INAV/MATEKF722PX 2.5.0 May 26 2020 / 12:47:41 (d84bcb733)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

Unknown601, 2020-06-02:
I have been trying to solve this for 2 days :-(      I'm at my wits' end!

giacomo892, 2020-06-02:
I have compass and GPS running on my F722PX without troubles.
It's likely you have alignment issues and interference.

Unknown601, 2020-06-02:
> I have compass and GPS running on my F722PX without troubles.
> It's likely you have alignment issues and interference.

I found it was been caused by my GPS been at a  -20 degree pitch angle.

Can I tell inav my gps/compass is tilted?

giacomo892, 2020-06-02:
In CLI:
align_board_roll 
align_board_pitch 
align_board_yaw

Next time do not open an issue here but contact the community on telegram or facebook

closed 2020-06-02 by giacomo892

Unknown601, 2020-06-02:
> In CLI:
> align_board_roll
> align_board_pitch
> align_board_yaw
> 
> Next time do not open an issue here but contact the community on telegram or facebook

Thanks,

Won't align_board change the FC alignment?