iNavFlight/inav #5804: Abnormal Flight Mode Detection - Fixed Wing

robthomson, 2020-06-02:
Inav works exceptionally well in amost all flight dynamics.

It can fail badly in a stall/spin scenario.

When flying in AIR MODE..  if you do go into a spin (intentional or not)  the flight controller will actively work against you in achieving recovery.

It will for example.. keep holding opposite control to bring the wing to level.  as there is no airflow over the surface; doing this is pointless and only serves to hold you in the stall/spin.

Currently the ONLY solution is is to engage manual mode and fix the problem yourself.

I believe it should be possible to enable a 'recovery mode' which can auto enable based on a couple of parameters.

1.  Rotation of Aircraft Greater than X degrees per second.
2.  Barometric Altitude falling greater than X speed.
3.  Air Speed (if pitot enabled)
4. Ground Speed less than X

I suspect just the first 2 would be needed.

So..  if you are rotating at greater than 20 degrees a second... and falling at greater than 0.5m per second...   auto enable manual mode for X seconds. or..  force all surface to neutral for X seconds

This would likely auto enable a pilot to determine best parameters for their model post some experimentation.  This would essentially mean the fly by wire would disable for a few seconds automatically if you are completely out of normal flight dynamics - but would enable the pilot commands to override and solve the problem.

brainbubblersbest, 2020-06-12:
Confirmed. 
Came in to a downwards corkscrew with full speed and the wing moves in other side of my stick direction. 
Do i need to mention that I'm a little frustrated right now? 😔
[image]

robthomson, 2020-06-25:
If its any consolation this issue caused me to loose a $2000 model. Yes.. frustrating!

digitalentity, 2020-06-25:
I totally understand the frustration. 

As far as I know no autopilot is able to recover from a spin without pilot's intervention. You can always switch to manual mode and attempt to recover yourself or configure the aircraft (rates and cruise throttle mostly) to make sure the stall doesn't happen.

We should indeed think about stall detection, but given the fact that we will have to collect data to figure what's common for stalls (which mean intentionally force our airplanes into this abnormal mode) it's unlikely we will even have this implemented.

robthomson, 2020-06-25:
To be fair..   its not that hard to gather info.

A stall or spin is not actauly a fatal issue - unless the flight controller is running in one of the stabilised modes.  In theory.. all it takes is the ability to stall and recover in 'manual mode' whilst flying los.

I would be happy to do data gathering?

Personally..  dont think its even needed to gather data.

I can trigger a basic detection via opentx using barometric decent + rotation.  This is just about enough to detect a spin.

The alternative way which totally ignores all off this is:

if { in stab mode } & {control response is not delivery rotation in proposed direction} & {delay > 3 seconds}{
   engage manual for 5 seconds;
}

At worst.. this will allow a 'blip' to manual to allow your own stick commands to override the 'stall'.

I am more than happy to be a sacrificial lamb to do the testing for this :-)

digitalentity, 2020-06-25:
Actually it's false positives I'm concerned about. If we want to make it an automatic thing - we want to have it more or less bullet-proof to avoid issues with unexpected  manual mode engagement.

brainbubblersbest, 2020-06-25:
I dont understand why the Autopilot can not detect the Rotation in relation to it's actual Position over the accelerometer? Nose is down? Or is here a Zero gravity in detection? Or what ever. Maybe anyone can explain?

robthomson, 2020-06-25:
Maybe the solution is to consider a slightly different flight mode.

Traditional MEMS stabilizers never had the issuer - air mode does.

The question then is..   what about a flight mode that is essentially full acro - but with a stabilizer?

So air mode.. without the fly by wire!

robthomson, 2020-06-25:
Another option may be to try and detect the buffeting that occurs just prior to a stall?  And then prevent you stalling in the first place!

closed 2024-02-02 by DzikuVx