iNavFlight/inav #5807: GPS not logging valid timestamp

Safranek42, 2020-06-03:
Hi,

I tried blackbox logging and then decode the logged data to .csv files with blackbox tools.

However, the two output file for gps data lacks any meaningful timestamp. In the csv file, the 'time (us)' field is constant zero. In the gps file, <time>2000-01-01T00:00:00.000000Z</time> is the timestamp in every row. On the other csv file there is correct time data.

Does anyone know why is this happening, or how can it be solved?

Thanks in advance for any help.

Gecete, 2020-06-14:
Same problem, I wanted to visualise the mission on google earth and after decoding blackbox I can just see the static image of the gps points, when hovering them I see all points have same timestamp 2000-01-01T00:00:00.000000Z so the animation cant be done by google earth

Dope-johnny, 2020-06-19:
I can confirm this issue with INAV 2.5.0

stronnag, 2020-06-19:
Works perfectly for me. It would help if you actually posted some useful information like CLI `diffs` and the BB logs.

digitalentity, 2020-06-19:
@Safranek42 please answer a few questions:

1. Did it ever work on earlier versions?
2. What GPS hardware are you using?
3. Do you see correct date/time in the OSD?
4. Please provide raw blackbox files with the problem and your configuration dump.

Dope-johnny, 2020-06-19:
@digitalentity I'll provide those infos as well:

1. I can not confirm it certainly because I don't have old logs - but I remember I got it working correctly earlier this year with INAV 2.3.0. I followed [pawels guide on YT](https://youtu.be/GEGywqPD2Uk)
2. Beitian BN-880
3. Yes
4. I did use cleanflight decoder v0.4.3-windows. Here I got the timestamp in the .gps.gpx file like @Gecete described. Now that you mentioned it I found the fork from inavflight - v0.4.4. With that I get the correct inital time in the .gps.gpx file but it doesn't change:
```
<?xml version="1.0" encoding="UTF-8"?>
<gpx creator="Blackbox flight data recorder" version="1.1" xmlns="http://www.topografix.com/GPX/1/1" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="http://www.topografix.com/GPX/1/1 http://www.topografix.com/GPX/1/1/gpx.xsd">
<metadata><name>Blackbox flight log</name></metadata>
<trk><name>Blackbox flight log</name><trkseg>
  <trkpt lat="48.7903512" lon="9.3760687"><ele>429</ele><time>2020-06-12T05:50:10.000000Z</time></trkpt>
  <trkpt lat="48.7903512" lon="9.3760687"><ele>429</ele><time>2020-06-12T05:50:10.000000Z</time></trkpt>
  <trkpt lat="48.7903511" lon="9.3760690"><ele>429</ele><time>2020-06-12T05:50:10.000000Z</time></trkpt>
...
  <trkpt lat="48.7897190" lon="9.3765041"><ele>426</ele><time>2020-06-12T05:50:10.000000Z</time></trkpt>
```
   With the --datetime option I get the correct (and changing) time in the .csv file:
```
loopIteration, time (us), dateTime, axisRate[0], axisRate[1], axisRate[2], axisP[0], axisP[1], axisP[2], axisI[0], axisI[1], axisI[2], axisD[0], axisD[1], axisD[2], fwAltP, fwAltI, fwAltD, fwAltOut, fwPosP, fwPosI, fwPosD, fwPosOut, rcData[0], rcData[1], rcData[2], rcData[3], rcCommand[0], rcCommand[1], rcCommand[2], rcCommand[3], vbat (V), amperage (A), BaroAlt (cm), rssi, gyroADC[0], gyroADC[1], gyroADC[2], accSmooth[0], accSmooth[1], accSmooth[2], attitude[0], attitude[1], attitude[2], debug[0], debug[1], debug[2], debug[3], debug[4], debug[5], debug[6], debug[7], motor[0], motor[1], servo[0], servo[1], servo[2], servo[3], servo[4], servo[5], servo[6], servo[7], servo[8], servo[9], servo[10], servo[11], servo[12], servo[13], servo[14], servo[15], navState, navFlags, navEPH, navEPV, navPos[0], navPos[1], navPos[2], navVel[0], navVel[1], navVel[2], navAcc[0], navAcc[1], navAcc[2], navTgtVel[0], navTgtVel[1], navTgtVel[2], navTgtPos[0], navTgtPos[1], navTgtPos[2], navSurf[0], energyCumulative (mAh), flightModeFlags (flags), stateFlags (flags), failsafePhase (flags), rxSignalReceived, rxFlightChannelsValid, hwHealthStatus, powerSupplyImpedance, sagCompensatedVBat, wind[0], wind[1], wind[2], IMUTemperature, baroTemperature, sens0Temp, sens1Temp, sens2Temp, sens3Temp, sens4Temp, sens5Temp, sens6Temp, sens7Temp, escRPM, escTemperature
  0, 328332637,2020-06-12T05:55:38.332637Z,   1,   6,   0,   0,   2,   0,   0,   0,   0,   3,  26,   0,   0,   0,   0,   0,   0,   0,   0,   0, 1496, 1500, 1500, 1000,   0, -283,   0, 1050, 16.930, 0.580, 244, 1023,   3,   1,  -2, 662, -31, 1956,  -5, -192, 1556, 1044,   0,   0, 72077,   0,   0,   0,   0, 1000, 1000, 1500, 1500, 1500, 1487, 1447, 1500, 1500, 1500, 1500, 1500, 1500, 1500, 1500, 1500, 1500, 1500,  26,   5, 106, 100, -1261, 298, -16,   0,   2,  -1,   5, -13,   5,   0,   0,   0,   0,   0,   0,   0, 0, ARM|ANGLE, GPS_FIX_HOME|GPS_FIX|SMALL_ANGLE|FIXED_WING, IDLE, 1, 1, 325, 0, 1680, 0, 0, 0, 308, 285, -1250, -1250, -1250, -1250, -1250, -1250, -1250, -1250, 134239981, 0
  1, 328333643,2020-06-12T05:55:38.333643Z,   1,   6,   0,   0,   2,   0,   0,   0,   0,   2,  26,   0,   0,   0,   0,   0,   0,   0,   0,   0, 1496, 1500, 1500, 1000,   0, -283,   0, 1050, 16.930, 0.580, 244, 1023,   2,   1,  -2, 662, -30, 1956,  -5, -192, 1556, 1044,   0,   0, 72077,   0,   0,   0,   0, 1000, 1000, 1500, 1500, 1500, 1487, 1447, 1500, 1500, 1500, 1500, 1500, 1500, 1500, 1500, 1500, 1500, 1500,  26,   5, 106, 100, -1261, 298, -16,   0,   2,  -1,   6, -13,   5,   0,   0,   0,   0,   0,   0,   0, 53, ARM|ANGLE, GPS_FIX_HOME|GPS_FIX|SMALL_ANGLE|FIXED_WING, IDLE, 1, 1, 325, 0, 1680, 0, 0, 0, 308, 285, -1250, -1250, -1250, -1250, -1250, -1250, -1250, -1250, 134239981, 0
  2, 328334648,2020-06-12T05:55:38.334648Z,   1,   6,   0,   0,   2,   0,   0,   0,   0,   2,  26,   0,   0,   0,   0,   0,   0,   0,   0,   0, 1496, 1500, 1500, 1000,   0, -283,   0, 1050, 16.930, 0.580, 244, 1023,   2,   1,  -2, 661, -29, 1957,  -5, -191, 1556, 1044,   0,   0, 72077,   0,   0,   0,   0, 1000, 1000, 1500, 1500, 1500, 1487, 1447, 1500, 1500, 1500, 1500, 1500, 1500, 1500, 1500, 1500, 1500, 1500,  26,   5, 106, 100, -1261, 298, -16,   0,   2,  -1,   6, -12,   5,   0,   0, 
...
944464, 1278231551,2020-06-12T06:11:28.231551Z,   0,   0,   0,   0,   0,   0, 152, -81, -163,   0,   0,   0,   0,   0,   0,   0,   0,   0,   0,   0, 1498, 1500, 1500, 1011,   0,   0,   0, 1049, 15.140, 0.730, 270, 534,   0,   0,   0, -671, 158, 1947,  73, 185, 802, 1045,   0,   0, 53925,   0,   0,   0,   0, 1000, 1000, 1500, 1500, 1500, 1474, 1338, 1500, 1500, 1500, 1500, 1500, 1500, 1500, 1500, 1500, 1500, 1500,   1,  37, 140, 100, -8399, 3551,  -6, -14, -17, -161, -81,  13,  -6,   0,   0,   0, -8035, -3837, 4562,   0, 1333, 0, GPS_FIX_HOME|GPS_FIX|SMALL_ANGLE|FIXED_WING, IDLE, 1, 1, 325, 36, 1515, -95, 450, 11, 268, 256, -1250, -1250, -1250, -1250, -1250, -1250, -1250, -1250, 134239981, 0
```
5. Log: https://drive.google.com/file/d/1iMAqs0hurnx-aQ_vF_dAgCg1PupWheUA/view?usp=sharing and dump: https://pastebin.com/z5KaGK9a

stronnag, 2020-06-19:
No problem. CSV has timestamps, GPX has timestamps.
```
blackbox_decode --merge-gps --datetime /LOG00001.TXT 
```

stronnag, 2020-06-19:
```
H Firmware revision:INAV 2.5.0 (694767938) MATEKF405SE
H Firmware date:Jun  7 2020 11:47:09
H Log start datetime:2020-06-12T05:50:10.719+00:00
H Craft name:KK_SCHIKK
H P interval:1/1
```
So 2020-06-12 07:50:10 CEST , nice way to spend a Friday morning.

[image]

Dope-johnny, 2020-06-19:
@stronnag I just used the exact same options to decode on 2 different windows machines and get the same .gpx files with static timestamps. Did you use the windows build of the balckbox decoder as well?

stronnag, 2020-06-19:
No. I avoid Windows like the plague. Arch Linux for the win!!

stronnag, 2020-06-19:
Wait a minute, in the GPX the timestamp doesn't change `<time>2020-06-12T05:50:10.000000Z</time>` (linux blackbox-decode).  I just checked against `2000-01-01T00:00:00.000000Z` as per first report, mea culpa. 

This is [blackbox-tools](https://github.com/iNavFlight/blackbox-tools) bug in generating the GPX, not a firmware bug as there are valid timestamps in the raw log, and the CSV with `--datetime`  has stamps from `2020-06-12T05:55:38.332637Z` to `2020-06-12T06:11:28.231551Z`. And there's a bug in the mwp bb replay summary (see image), it gets the elapsed time wrong (ignores the start offset).

stronnag, 2020-06-19:
@Dope-johnny  Did `blackbox-decode` ever generate sequential time stamps in GPX? 
Even if this is a legacy mis-feature, it should be fixed. Consider me on the case.

stronnag, 2020-06-19:
To answer my own question. Yes it did, in my log stash I have a BBL GPX from  2017-12-30 that has sequential GPX timestamps. A [blackbox-tools](https://github.com/iNavFlight/blackbox-tools) regression to be fixed.

stronnag, 2020-06-19:
it will be interesting to see @Safranek42's logs to see if this is the same problem.

stronnag, 2020-06-19:
Oh, wait again. This is not an mwp bbl replay bug, it's the horrible `blackbox_decode` "duplicate rows with null time bug" Ouch.
[image]

stronnag, 2020-06-19:
And regardless of the bogus time-stamp bug, the recording of GPS timestamps mystery is at least solved.

```
 // If we're not logging every loop iteration, we include a timestamp field in the GPS frame:
```

My logs are not every iteration, I get GPS timestamps in the GPX, @Dope-johnny 's logs are every iteration, so they don't. It's a feature, but one we can probably improve.

digitalentity, 2020-06-19:
@stronnag looks like you have a decent idea what's going on here, thanks for looking!
In any case this issue doesn't belong to this repository - moving to blackbox-tools repo.