iNavFlight/inav #5809: Craftname displayed on "armed" screen

AnandNL, 2020-06-04:
## Current Behavior

On the armed screen there is the home point and time/date, having the craftname constantly in the osd is pointless for me.
## Desired Behavior

Added [craftname] to that screen only shown on ARM
## Suggested Solution

## Who does this impact? Who is this for?

## Additional context

I use a standard layout for all my planes and that makes it hard to see the difference between flights with different aircraft when i'm at home reviewing older flights.

Mateyhv, 2020-06-04:
Why don't you deselect the name from the OSD menu? Just that simple

AnandNL, 2020-06-04:
[image]
 I want it in that screen, just when i arm, i don't use it in the normal osd

Mateyhv, 2020-06-04:
Ok, your initial post was not clear enough. So you want the name on that momentary screen that appears for a second when you arm the model.

AnandNL, 2020-06-04:
> Ok, your initial post was not clear enough. So you want the name on that momentary screen that appears for a second when you arm the model. You should tag the thread as "Feature request"

i tried that, got the preformatted template but not the feature request sticker

craigmunday, 2020-07-22:
I created a branch to introduce this change.  The example screens are below.

When the craft name is set in the CLI with the command "set name=MY CRAFT NAME"
[image]

And this screen when the craft name is not set (i.e. "set name="
[image]

I will create a pull request if there is agreement that these screens are OK.

AnandNL, 2020-07-22:
Thats great

closed 2020-07-29 by digitalentity