iNavFlight/inav #5812: Logic Condition Format 2.5 RC2 ?

breadoven, 2020-06-06:
When trying to restore settings from 2.4.0 to 2.5.0 RC2 the CLI throws a Parse error for the Logic Conditions.

2.4.0 Diff for Logic Conditions:
**# logic
logic 0 1 3 1 6 0 1720 0**

After manually reentering the above Logic Condition into 2.5.0 RC2 via the Configurator Programming Tab and saving, the CLI now shows the following:
**# logic
logic 0 1 -1 3 1 6 0 1720 0**

Is this a bug with the "-1" or just an as yet undocumented change ?

Apologies if this has already been covered before, couldn't find anything obvious.

Thanks !

DzikuVx, 2020-06-08:
Yes, it changed. The new format is described here https://github.com/iNavFlight/inav/blob/master/docs/Logic%20Conditions.md

closed 2020-06-08 by DzikuVx

haiduc32, 2020-09-20:
https://github.com/iNavFlight/inav/blob/master/docs/Logic%20Conditions.md is missing.

stronnag, 2020-09-20:
subsumed into https://github.com/iNavFlight/inav/blob/master/docs/Programming%20Framework.md