iNavFlight/inav #5817: Does Mag pitch angle matter on multirotor?

ShawnzAZ, 2020-06-07:
In the early GPS enabled multirotors people would put the GPS / compass on a mast that stood perpendicular to frame / flight contorller.

As Inav is now finding it's way to smaller / faster quads and GPS 'tails' are popular.   Many if not most of these  have the GPS/MAG tilted 'back' 30 degrees or so.

Yeah, we calibrate it in this position, but do we need to let the FC know that the mag is not aligned to the FC, similar to board pitch alignment for the FC
[image]
?

ShawnzAZ, 2020-06-07:
Currently using INav 2.5 RC2 on Matek F722-PX & Matek M8Q-5883 GPS/MAG

# version
# INAV/MATEKF722PX 2.5.0 May 26 2020 / 12:47:41 (d84bcb733)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000 -0.810  0.659 -1.000
mmix 1  1.000 -1.000 -0.659  1.000
mmix 2  1.000  0.810  0.659  1.000
mmix 3  1.000  1.000 -0.659 -1.000

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100
servo 8 1000 2000 1500 100
servo 9 1000 2000 1500 100
servo 10 1000 2000 1500 100
servo 11 1000 2000 1500 100
servo 12 1000 2000 1500 100
servo 13 1000 2000 1500 100
servo 14 1000 2000 1500 100
servo 15 1000 2000 1500 100

# logic
logic 0 0 -1 0 0 0 0 0 0
logic 1 0 -1 0 0 0 0 0 0
logic 2 0 -1 0 0 0 0 0 0
logic 3 0 -1 0 0 0 0 0 0
logic 4 0 -1 0 0 0 0 0 0
logic 5 0 -1 0 0 0 0 0 0
logic 6 0 -1 0 0 0 0 0 0
logic 7 0 -1 0 0 0 0 0 0
logic 8 0 -1 0 0 0 0 0 0
logic 9 0 -1 0 0 0 0 0 0
logic 10 0 -1 0 0 0 0 0 0
logic 11 0 -1 0 0 0 0 0 0
logic 12 0 -1 0 0 0 0 0 0
logic 13 0 -1 0 0 0 0 0 0
logic 14 0 -1 0 0 0 0 0 0
logic 15 0 -1 0 0 0 0 0 0

# gvar
gvar 0 0 -32768 32767
gvar 1 0 -32768 32767
gvar 2 0 -32768 32767
gvar 3 0 -32768 32767

# gf
gf 0 0 -1 0 0 0 0
gf 1 0 -1 0 0 0 0
gf 2 0 -1 0 0 0 0
gf 3 0 -1 0 0 0 0
[... 1159 lines cut ...]
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 60
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 2200
set battery_capacity_warning = 660
set battery_capacity_critical = 440
set battery_capacity_unit = MAH

# end the command batch
batch end

Beardyblair, 2020-06-08:
I have the exact same setup as you describe. In short, no pitch does not matter.

FYI. I was unable to get a reliable compass reading from the Matek GPS/Compass with it mounted at the rear like that. I moved it to a post and it worked well. Are you having compass issues?

ShawnzAZ, 2020-06-11:
the heading graph seems to work perfect.   What I'm seeing is if I land / pop up in a hover and set position hold it will hold position perfect.

If I transition to poshold from forward flight, or even if I command movement in position hold it's like the gain for whatever is commanding to the corrections isn't strong enough and it even tually toilet bowls until I have to just flip to acro and recover.

I don't think it is 100% a mag issue or why would it hold from a clean hover?   It definately seems tune related.   As I said before I will do a compass calibration in the field (vs in my living room) for good measure.   I was thinking maybe the mag being inclined was a factor in the readings since when people put them on a post like the old days it's now also usually perpendicular to the frame and in alignment with the FC.

brainbubblersbest, 2020-06-12:
it is 100% a mag issue. And yes. Tilt needs to be configured and after that, calibrate.
In Sensors tab a perfekt aligned and calibrate compass had These values. 
Leveled= z axis in negative, must have same value as flipped over in positive
Tilted 90degree to Front = x axis in negative, must have same value as tilted 90 degree backwards in positive
Tilted 90 degree to left = y axis in negative, must have same value as tilted 90 degree to the right in positive.

And the values must reach it's Max at this angles.
If Max values are reached not in 90degree but at 120 degree or 60 degree. 
So this is caused by a 30degree physical rotated magnetometer.
90-30=60
90+30=120

ShawnzAZ, 2020-06-12:
> it is 100% a mag issue. And yes. Tilt needs to be configured and after that, calibrate.
> In Sensors tab a perfekt aligned and calibrate compass had These values.
> Leveled= z axis in negative, must [...]

So mine is installed with the arrow facing rear so it ends up 90 CW flip.

Will the pitch alignment be done 100% in CLI?   I don't remember seeing pitch angle for the mag in the gui.   Is there a place to see the current mag pitch angle?  maybe in OSD?

I feel a feature request coming on :)

ShawnzAZ, 2020-06-12:
Looks like 'ALIGN_MAG_Pitch' will be where to tweak.

So what value = 30 degree tilted back?   -300?

Guess this will be the place to look:
[image]

# get mag
align_mag = CW90FLIP
Allowed values: DEFAULT, CW0, CW90, CW180, CW270, CW0FLIP, CW90FLIP, CW180FLIP, CW270FLIP

mag_hardware = QMC5883
Allowed values: NONE, AUTO, HMC5883, AK8975, GPSMAG, MAG3110, AK8963, IST8310, QMC5883, MPU9250, IST8308, LIS3MDL, FAKE

mag_declination = 0
Allowed range: -18000 - 18000

magzero_x = -563
Allowed range: -32768 - 32767

magzero_y = 357
Allowed range: -32768 - 32767

magzero_z = 15
Allowed range: -32768 - 32767

mag_calibration_time = 30
Allowed range: 30 - 120

align_mag_roll = 0
Allowed range: -1800 - 3600

align_mag_pitch = 0
Allowed range: -1800 - 3600

align_mag_yaw = 0
Allowed range: -1800 - 3600

imu_dcm_kp_mag = 10000
Allowed range: 0 - 65535

imu_dcm_ki_mag = 0
Allowed range: 0 - 65535

inav_auto_mag_decl = ON
Allowed values: OFF, ON

brainbubblersbest, 2020-06-12:
Try
align_mag_roll = 300
align_mag_pitch = 1800
align_mag_yaw = 900

If it's still weird, try align_mag_roll = - 300

brainbubblersbest, 2020-06-13:
I would apreciate any Feedback 😉

closed 2020-08-31 by stale[bot]