iNavFlight/inav #5818: RPM telemetry (DShot enabled) & RPM filter broken under 2.5.0 stable release

leeph, 2020-06-07:
## Current Behavior
Two separate fixed-wing craft with BLHeli_32 ESCs with telemetry wires connected, under 2.4 telemetry was working fine, including RPM filter (RPM filter task present in scheduler list).

## Steps to Reproduce
After upgrading to 2.5 stable with a merged diff, RPM telemetry is broken entirely - no RPM reporting through debug_mode = erpm and RPM filter task not present in scheduler list, even though it is enabled.

## Expected behavior
RPM telemetry should continue to work under 2.5.0, but it appears it is broken, at least under fixed-wing platform.

## Additional context
The two craft are running Matek F722-Wing and F722-STD respectively, with BLHeli_32 ESCs with telemetry wire connected to RX pin (hardware UART, defined correctly under ports tab).

Please find diff all for the 2.5.0 config at: https://pastebin.com/hBHUXWq6

---

- FC Board name and vendor: MatekSys F722-Wing, F722-STD
- INAV version string: INAV/MATEKF722SE 2.5.0 Jun  7 2020 / 11:43:20 (694767938)

Mateyhv, 2020-06-08:
Same here on Matek F411 Wing, tested it on Softserial2 and worked nicely under 2.5 RC2, switched to 2.5 stable and ESC telemetry is gone.

Mateyhv, 2020-06-17:
Tested after releasing v2.5.1. Works nicely! Thanks

leeph, 2020-06-17:
Yes thanks for fixing this regression guys, 2.5.1 is now a really solid release for me. Would love to see bidirectional DShot at some stage though!

PR #5825 (2020-06-08): Esc telemetry fix
Fixes #5818
closed 2020-06-17 by giacomo892