iNavFlight/inav #5819: 5 Inch Mini Quad is wobbeling on its way home

Burnst81, 2020-06-07:
Hello INAV People

im trying hard to get more and more into it. To get more and more variables and commands but its hard to get all of them on my own.

I build an 5 Inch mini Quad with INAV 2.5. It performs very good! I love it. Althold and Position hold work perfect out of the box. Also RTH works fine exept one thing. The copter is wobbeling up and down and sometimes do big bounces on its way home. 

There are so many variables i read. For Wings.. Multirotors and some for both? .. i don´t know which setting i should play with to get it fixed.

i uploaded a Video to let you see what i mean. At 0:08 min i turn the RTH switch. It is in RTH the whole rest of the Video. You can see the wobbeling and those bounces from time to time.

[https://youtu.be/0McS80DIa6o](https://youtu.be/0McS80DIa6o)

Matek F722 SE
Spedix 30A ESC
DJI Air Unit
T-Motor Pacer 1750kv 2207

my diff /all

diff /all

# version
# INAV/MATEKF722SE 2.5.0 May 26 2020 / 12:36:10 (d84bcb733)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix

# servo

# logic

# gvar

# gf

# feature
feature GPS
feature PWM_OUTPUT_ENABLE

# beeper

# map

# serial
serial 0 2 115200 57600 0 115200
serial 2 2097152 115200 115200 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1800 2100
aux 1 1 1 1350 1450
aux 2 2 1 1150 1250
aux 3 3 2 1300 1700
aux 4 11 2 1700 2100
aux 5 10 3 1300 2100
aux 6 30 4 1900 2100
aux 7 13 1 1950 2050

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 13 13 V
osd_layout 0 1 12 2 H
osd_layout 0 3 8 6 H
osd_layout 0 4 8 6 H
osd_layout 0 7 8 15 V
osd_layout 0 9 12 14 V
osd_layout 0 11 13 15 V
osd_layout 0 12 21 1 V
osd_layout 0 13 26 5 V
osd_layout 0 14 1 1 V
osd_layout 0 15 23 6 V
osd_layout 0 22 12 1 V
osd_layout 0 23 14 1 V
osd_layout 0 24 12 6 H
osd_layout 0 26 26 7 V
osd_layout 0 28 23 14 H
osd_layout 0 30 1 12 V
osd_layout 0 32 17 15 V
osd_layout 0 34 10 8 H
osd_layout 0 50 12 2 H
osd_layout 0 52 12 3 H
osd_layout 0 99 13 8 H

# master
set looptime = 500
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 120
set gyro_lpf_type = PT1
set dynamic_gyro_notch_enabled = ON
set dynamic_gyro_notch_q = 250
set dynamic_gyro_notch_min_hz = 120
set acc_hardware = MPU6000
set acc_lpf_hz = 102
set acczero_x = 49
set acczero_y = 25
set acczero_z = 119
set accgain_x = 4089
set accgain_z = 4028
set align_mag = CW270FLIP
set mag_hardware = QMC5883
set mag_declination = 230
set magzero_x = -383
set magzero_y = 12
set magzero_z = -398
set baro_hardware = BMP280
set pitot_hardware = ADC
set rssi_channel = 12
set rssi_min = 49
set serialrx_provider = CRSF
set blackbox_rate_denom = 2
set motor_pwm_rate = 8000
set motor_pwm_protocol = DSHOT300
set throttle_idle =  6.000
set failsafe_procedure = RTH
set current_meter_scale = 206
set model_preview_type = 3
set servo_pwm_rate = 330
set small_angle = 45
set applied_defaults = 2
set gps_sbas_mode = EGNOS
set gps_ublox_use_galileo = ON
set mc_airmode_type = THROTTLE_THRESHOLD
set inav_auto_mag_decl = OFF
set nav_disarm_on_landing = ON
set nav_auto_speed = 1389
set nav_auto_climb_rate = 400
set nav_rth_abort_threshold = 30000
set nav_rth_altitude = 5000
set nav_mc_bank_angle = 35
set debug_mode = GYRO
set name = Flying Dutchman

# profile
profile 1

set mc_p_pitch = 58
set mc_i_pitch = 65
set mc_d_pitch = 35
set mc_p_roll = 54
set mc_i_roll = 60
set mc_d_roll = 35
set mc_p_yaw = 35
set mc_i_yaw = 80
set mc_p_level = 35
set max_angle_inclination_rll = 400
set max_angle_inclination_pit = 400
set dterm_lpf_hz = 110
set dterm_lpf_type = PT1
set dterm_lpf2_hz = 170
set dterm_lpf2_type = PT1
set use_dterm_fir_filter = OFF
set dterm_setpoint_weight =  0.750
set nav_fw_pos_z_i = 10
set d_boost_factor =  1.500
set antigravity_gain =  5.000
set antigravity_accelerator =  5.000
set tpa_rate = 20
set tpa_breakpoint = 1650
set rc_yaw_expo = 70
set roll_rate = 100
set pitch_rate = 100
set yaw_rate = 90

# battery_profile
battery_profile 1

# end the command batch
batch end

#

DzikuVx, 2020-06-11:
You can:
* lower `mc_p_level`
* lower `nav_mc_vel_xy_p`
* lower `nav_auto_speed `

closed 2020-06-11 by DzikuVx

Burnst81, 2020-06-12:
thanks a lot! I will try it!