iNavFlight/inav #5820: 2.5 Reversible DSHOT Throttle Is Inverted

KenImhof, 2020-06-07:
## Current Behavior

With DSHOT protocol and Reversible Motors, the reverse throttle output is inverted.  The idle throttle just below center stick outputs full reverse throttle and full stick down outputs idle throttle.

## Steps to Reproduce

1. Flash version 2.5 with full erase.
2.Enable Reversible Motors with DSHOT protocol.
3. Arm controller and move throttle just below center stick, which outputs full reverse throttle (1000us).

## Expected behavior

1. Full reverse throttle 1000Us should output when the throttle stick is in the full down position and NOT near the center throttle position.

## Suggested solution(s)

In the writeMotors(void) function for the DSHOT Reverse call to the handleOutputScaling call, swap the throttleRangeMin and throttleRangeMax values.

digitalentity, 2020-06-08:
@DzikuVx this could be related to rover/boat support and reversible motor mixer refactoring.

KenImhof, 2020-06-08:
Refactoring looks excellent and very well done!  I suspect it just was not verified with DSHOT protocol.

floflolapin, 2021-04-13:
Hi. I have the same issue. I changed my esc for blheli32 ones thinking this could be the problem and updated to 2.6.1 but it's still the same.
It work as intended with standard pwm protocol though...

closed 2022-02-23 by DzikuVx