iNavFlight/inav #5821: Matek 722SE Sbus not working

UnCorpuscler, 2020-06-08:
went from 2.4 to 2.5 and it can no longer communicate with my r9mm, reverted to 2.4 to confirm this and is working properly.

digitalentity, 2020-06-08:
Can't reproduce. Please flash with full erase and reconfigure.

DzikuVx, 2020-06-09:
I have 2 quads with this FC, both with SBUS and INAv 2.5. On both works without any problems

closed 2020-06-09 by DzikuVx

UnCorpuscler, 2020-06-09:
I just reproduced this on the matek F405SE as well, same deal, flash 2.5 no channel response, flash 2.4, back to normal.

digitalentity, 2020-06-09:
As advised in the release notes, you should flash with full erase. I also have F405-WING and S.Bus works perfectly

UnCorpuscler, 2020-06-09:
@digitalentity I thought that was obvious, I do that on every flash. I’m tellin ya mate I’m not dull, if I leave everything as is and flash 2.5 there’s nothing. However because no one else is getting this, I will leave it be.

digitalentity, 2020-06-09:
@UnCorpuscler this is interesting. It's not the first time FrSky is breaking S.Bus compatibility. Can you try with another receiver, i.e. XM+?