iNavFlight/inav #5828: Cant get past 40% throttle in configurator or when connected with RC controller?

IrnFru, 2020-06-08:
In the iNav configurator or connected with RC controller I cant get my throttle value past 40%.
 It happened on the maiden with iNav 2.5 rc2 , tried to find the problem myself without any success then noticed that 2.5 stable had been realesed so re flashed and still have the same issue? 

****
## Steps to Reproduce
1. Drag throttle bar in outputs tab
2. Arm on the ground the throttle up with RC controller

- Matek 722 Wing
- INAV version string :2.5.0

Mateyhv, 2020-06-08:
Definitely that doesn't seem to be related to Inav, You should check your RC system which by the way you didn't even mention.

IrnFru, 2020-06-08:
> Definitely that doesn't seem to be related to Inav, You should check your RC system which by the way you didn't even mention.

Sorry 

> Definitely that doesn't seem to be related to Inav, You should check your RC system which by the way you forgot to mention.

Hi ,
 I've updated my query to include my RC system 

  It definitly happens in the iNav outputs page, not just with my RC system which I did mention.
Thanks.

IrnFru, 2020-06-09:
Turns out it was a dud ESC straight out the box,  thanks sonicmodel

closed 2020-06-09 by IrnFru