iNavFlight/inav #5833: GPS Satelite Count - DJI FPV system

SweetBear1, 2020-06-09:
I'm not sure if this is a bug, or a problem with the setup inputs on my end (my mistake in configurator). I have a Matek F405 setup using INAV 2.4 with a BN220 GPS and DJI system is properly showing OSD information (the ones that are available). Everything seems to be working with the exception of GPS satellite count data. For some reason it always shows 14 satellites which I know is incorrect from plugging the FC into my computer and checking on the INAV configurator. I also have a Matek F405 using a Matek GPS receiver experiencing the same issue when using DJI FPV. Someone told me their betaflight setup with DJI FPV shows 14 GPS satellites to start, then goes to 0, then shows an accurate count. I havent tested anything in betaflight to confirm this myself.

digitalentity, 2020-06-09:
This is a known issue on DJI HD side, they show the incorrect number of satellites until there is a GPS fix. Does the number of satellites still show incorrect once you have a GPS fix?

SweetBear1, 2020-06-10:
> This is a known issue on DJI HD side, they show the incorrect number of satellites until there is a GPS fix. Does the number of satellites still show incorrect once you have a GPS fix?

Thank you for explaining this! I don't know how I just never saw it switch in the goggles, and I didn't see this explained anywhere for INAV. I tested it a few minutes ago and I can confirm that after a few minutes, it changed from 14 to 4....5....6.....8 satellites. Sorry for opening an issue for this, hopefully others will find this information useful. Thanks for the help!

closed 2020-06-10 by SweetBear1

patkszen, 2020-09-11:
Mine is showing incorrect satellite count after I get GPS fix. It was showing 14 when my GPS was not working due to issues with the module itself.

nullx8, 2021-06-01:
> This is a known issue on DJI HD side, they show the incorrect number of satellites until there is a GPS fix. Does the number of satellites still show incorrect once you have a GPS fix?

is there away of knowing the fix is still pending ? i just noticed in one flight where I simply never had a fix under clear sky, I assume this is a OSD issue (but can't be sure)