iNavFlight/inav #5835: Doesn't detect gyro and accell

FFF200, 2020-06-09:
OMNIBUS F4 Nano V6
Target: Firework V2

INav 2.0-2.4 works well.
INav 2.5 - does not define an accelerometer and gyro.
In Configuration, resets it to NONE.
Manual selection doesn't help.
In the upper panel, accel and gyro are lit red.
Flash witch full chip erase.

[image]

[image]

[image]

Set manually. Does not work.

[image]

[image]

Again I flash with full Erase **iNav 2.4** - and everything works.
On iNav 2.2, I flew safely on this setup until I decided to upgrade to 2.5 to get new filters.

iNav 2.4-immediately after the firmware:

[image]

# diff

# version
# INAV/FIREWORKSV2 2.5.0 Jun  7 2020 / 11:10:57 (694767938)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

# start the command batch
batch start

# resources

# mixer

# servo mix

# servo

# logic

# gvar

# gf

# feature

# beeper

# map

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout

# master
set looptime = 500
set gyro_sync = OFF
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 110
set gyro_lpf_type = PT1
set dynamic_gyro_notch_enabled = ON
set dynamic_gyro_notch_q = 250
set dynamic_gyro_notch_min_hz = 120
set acc_hardware = MPU6000
set mag_hardware = QMC5883
set baro_hardware = BMP280
set pitot_hardware = NONE
set applied_defaults = 2
set mc_airmode_type = THROTTLE_THRESHOLD

# profile
profile 1

set mc_p_pitch = 44
set mc_i_pitch = 75
set mc_d_pitch = 25
set mc_i_roll = 60
set mc_p_yaw = 35
set mc_i_yaw = 80
set dterm_lpf_hz = 110
set dterm_lpf_type = PT1
set dterm_lpf2_hz = 170
set dterm_lpf2_type = PT1
set use_dterm_fir_filter = OFF
set dterm_setpoint_weight =  0.750
set d_boost_factor =  1.500
set antigravity_gain =  2.000
set antigravity_accelerator =  5.000
set tpa_rate = 20
set tpa_breakpoint = 1200
set rc_yaw_expo = 70
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 60

# battery_profile
battery_profile 1

# end the command batch
batch end

FFF200, 2020-06-10:
The main and only gyroscope is not defined...

In iNav 2.4, it was registered separately, there is no description of it here, or rather it stands as secondary in the conditions target = FIREWORKS V2

[image]

digitalentity, 2020-06-10:
Interesting. This change was tested on FIREWORKS V2 board. Can you please post output of the status and version command from 2.4.0 ?

FFF200, 2020-06-10:
Please write what you need to output.
I will do.

Now I will compile the firmware with my changes, check it in the evening and test it. But it will only be for the OMNIBUS F4 Nano V6.
In IMU_1, I will write the MPU6000 parameters from iNav 2.4

In 2.4, there was an explicit definition of MPU6000 and it worked.
In 2.5, it was removed and compatibility with OMNIBUS F4 Nano V6 by Target disappeared.

FFF200, 2020-06-10:
status and version - ОК.
I'll write this evening.

FFF200, 2020-06-10:
I'll test it tonight:

#if defined(OMNIBUSF4V6)
#   define IMU_1_CS_PIN            PA4
#   define IMU_1_SPI_BUS           BUS_SPI1
#   define IMU_1_ALIGN             CW180_DEG
#   define IMU_2_CS_PIN            PC14
#   define IMU_2_SPI_BUS           BUS_SPI1
#   define IMU_2_ALIGN             CW0_DEG
#else
    // FIREWORKS V2 - only for OMNIBUS F4 Nano V6
#   define IMU_1_CS_PIN            PA4			// OmnibusF4 Nano v6 and OmnibusF4 V6 has a MPU6000
#   define IMU_1_SPI_BUS           BUS_SPI1		// OmnibusF4 Nano v6 and OmnibusF4 V6 has a MPU6000
#   define IMU_1_ALIGN             CW180_DEG	// OmnibusF4 Nano v6 and OmnibusF4 V6 has a MPU6000
#   define IMU_2_CS_PIN            PC14  		// replaced from OMNIBUSF4V6, does not interfere with work, tested flashing OMNIBUSF4V6  
#   define IMU_2_SPI_BUS           BUS_SPI1		// replaced from OMNIBUSF4V6, does not interfere with work, tested flashing OMNIBUSF4V6  
#   define IMU_2_ALIGN             CW0_DEG		// replaced from OMNIBUSF4V6, does not interfere with work, tested flashing OMNIBUSF4V6  
#endif

digitalentity, 2020-06-11:
Fix verified on Omnibus F4 Nano V6

closed 2020-06-11 by digitalentity

markpmasonst, 2020-06-11:
I don't believe this is fixed. The final inav 2.5 still has the issue.

digitalentity, 2020-06-11:
2.5.1 containing the fix is not released yet. Wait for it

AlexAlle2020, 2020-06-12:
"Temporary solution. CLI: set gyro_to_use = 1"  Did it and it works. How reliable is this solution, can I fly far?

FFF200, 2020-06-12:
A colleague checked and flies. Please note that the gyroscope will need to be turned over in the settings, since the second gyroscope is used from the fireworks.  the fix is not quite correct in version 2.5.1. it will work exactly the same way - with an inverted gyroscope. in omnibus nano, it is located as in omnibusf4. in 2.4, this was taken into account.

AlexAlle2020, 2020-06-12:
да конечно я перевернул , полетал на коптере пару паков рядом (пару часов) , так вроде все ок . Глвное чтоб гира где то далеко не отвалилась)

AlexAlle2020, 2020-06-12:
I also noticed that the current sensor does not read from the telemetry of the motor regulators, and the video transmitter power is not regulated.

FFF200, 2020-06-12:
> да конечно я перевернул , полетал на коптере пару паков рядом (пару часов) , так вроде все ок . Глвное чтоб гира где то далеко не отвалилась) спасибо.

Удачи!

FFF200, 2020-06-12:
> I also noticed that the current sensor does not read from the telemetry of the motor regulators, and the video transmitter power is not regulated.

у меня регуляторы blhelly s без телеметрии. поэтому использую внешний датчик на RX4.

FFF200, 2020-06-12:
> I also noticed that the current sensor does not read from the telemetry of the motor regulators, and the video transmitter power is not regulated.

I have blhelly s controllers without telemetry. therefore, I use an external sensor on the RX4. I will check the video transmitter control. but the pit stop seems to be working

Elaugaste, 2020-06-14:
Current via ESC telemetry not work on nano v6 :(

FFF200, 2020-06-14:
Analog current on rx4:

-#define UART4_RX_PIN            PA1
+#define UART4_RX_PIN            NONE

-#define ADC_CHANNEL_1_PIN               PC1
+#define ADC_CHANNEL_1_PIN               PA1

Profit.

AlexAlle2020, 2020-06-19:
> > да конечно я перевернул , полетал на коптере пару паков рядом (пару часов) , так вроде все ок . Глвное чтоб гира где то далеко не отвалилась) спасибо.
> 
> Удачи!

Я думаю меня простят что пишу на русском языке , в инглише не силен . А боюсь что гугл переведет не верно . OMNIBUS F4 Nano V6 Target: Firework V2, USB смотрит назад  . С выходом Inav 2.5.1  я заметил что все так же аксель надо разворачивать не так как в Inav 2.3.0 положение другое Pitch Degrees 180 , но это не беда , развернул чтоб модель в Inаv конфигураторе вращалась верно . Танцы начались с включением компаса , раньше положение компаса было cw90 flip в inav 2.3.0 и летало все нормально   , но такое положение никак не  подходит для inav 2.5.1  , чтоб моделька в конфигураторе себя вела адекватно ,пришлось поставить MAG AlignmentGPS CW90 .но с таким расположением коптер при удержании точки по ЖПС улетает в китай, компас конечно калибровал по новой  . Я понимаю что проблема в расположении акселя и компаса , но ведь моделька во вкладке Setup ведет себя адекватно . ХЗ куда копать .

Dteksolutions, 2022-09-05:
I'm having the exact same issue on 5.1.0, 5.0 etc.

willyfpv, 2024-12-06:
I have the same issue trying to update to 7.1. can you please fix this bug?

sensei-hacker, 2024-12-07:
> I have the same issue trying to update to 7.1. can you please fix this bug?

If the group and accelerometer aren't detected, that generally means you flashed a target that is not correct for the FC you have.

Please attach a photo of your FC and a screenshot of the top of Configurator, where it shows which target has been flashed.

willyfpv, 2024-12-08:
i know is the correct target because reverting to 2.0.1 all works perfectly.
btw
[image]
[image]
ad the status command on cli report this:

INAV/FIREWORKSV2 7.1.2 May 31 2024 / 22:08:14 (4e1e59eb)
GCC-10.3.1 20210824 (release)
System Uptime: 21 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 0.03V (0S battery - NOT PRESENT)
CPU Clock=168MHz, BARO=BMP280
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=UNAVAILABLE, ACC=NONE, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
Stack size: 6144, Stack address: 0x10010000, Heap available: 1648
I2C Errors: 0, config size: 10253, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 2, used = ADC 2
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 1, used = ADC 1
  AIRSPEED : configured = none, used = none
System load: 54, cycle time: 1015, PID rate: 985, RX rate: 9, System rate: 9
Arming disabled flags: ANGLE CAL HWFAIL RX CLI
OSD: MAX7456 [30 x 16]
VTX: band: A, chan: 2, power: 1 (25 mW), freq: 5845 MHz

willyfpv, 2024-12-08:
thank you for the support!!!

sensei-hacker, 2024-12-08:
@willyfpv 
You posted in an old thread about the Nano V6, saying you have the same FC. Which works fine with INAV 7.2.1 - I have one right here.

When I asked for a picture of your board, you posted the instruction sheet for a Fireworks V2.

Please post a pic of YOUR FC.
I can't make your FC work without knowing what YOU have. A Fireworks V2 is not the same board board as a Nano V6, is not the same as Nano V7, is not the same as an Omnibus V5.

sensei-hacker, 2024-12-08:
For reference, here's what I mean.
Here are pics of my Nano V6 rev 2. (Running INAV 7.1.2).

[image]

[image]

willyfpv, 2024-12-08:
> @willyfpv You posted in an old thread about the _Nano V6_, saying you have the same. (Which works fine with INAV 7.2.1 - I have one right here).
> 
> When I asked for a picture of your board, you posted the instruction sheet for a _Fireworks V2_.
> 
> Please post a pic of YOUR FC. I can't make your FC work without knowing what YOU have. A Fireworks V2 is not the same board as a Nano V6, is not the same as Nano V7, is not the same as an Omnibus V5. Please post a good pic of YOUR flight controller.
> 
> Thank you.

I'm so sorry,my bad ,my FC IS THE V2,the one in the instructions sheet.
I cannot post a real pic because is inside a FUNJET under a vtx.
I mistakenly thought that the target was the only thing that was important.
I'm new to interacting in the thread of GitHub🙏so sorry

sensei-hacker, 2024-12-08:
> I'm so sorry,my bad ,my FC IS THE V2,the one in the instructions sheet. I cannot post a real pic because is inside a FUNJET under a vtx. I mistakenly thought that the target was the only thing that was important. I'm new to interacting in the thread of GitHub🙏so sorry

Okay, likely you will have better luck if you go into the Configuration tab of Configurator and select MPU6000 for your accelerometer. The default value of "auto" may also work.  Please let me know if that solves your issue.

If not, we may have to find a way to get a close-up pic of your board - even though that would be a hassle.

willyfpv, 2024-12-09:
[image]
this is the board.
ive tried your solution but with no luck,
after messing a little around with github i've found a solution that works (only tested on ground )

this commands : 
set gyro_to_use = 1
set acc_hardware = auto
and magically now detects both...any idea why? i want to understand a bit more if you can explain it to me.
THANKS a lot

sensei-hacker, 2024-12-10:
Glad you got that working.

That's what the manufacturer says to do for a related board, the Omnibus V6:

https://store.myairbot.com/flight-controller/omnibusf4v6.html

I was thinking selecting it on the Configuration tab would have the same effect, but you said you tried that and it didn't.

I've updated the documentation for these boards:
https://github.com/iNavFlight/inav/blob/96078ad1a7bf2aeac48150c7a017ec60f4367c3c/docs/boards/Omnibus%20F4.md

willyfpv, 2024-12-10:
Thank you for the bottom of my heart for the patience and the support in resolving this
Has been a pleasure meeting you.
Thank you