iNavFlight/inav #5837: Esc Temperature and RPM not working in INav 2.5

Tduru, 2020-06-09:
****
## Current Behavior
Right I can confirm that RPM and Temperature for ESC telemetry on INav 2.5 is somewhat broken. At first I thought my esc was faulty (my S1100 harrier is running a Racerstar tatto 52a BHeli32 on 6S. FC: Matek F405 Wing) so I tested my other plane which was fitted which was fitted with a new Holybro tekko32 35a esc and is running on 4S. The FC I'm using is a F765 Wing.

Before I flashed the board with INav 2.5, I tested the rpm via debug mode and it was working. My osd also showed the rpm values.

After I flashed to INav 2.5, not only does RPM and Temperature not work (Not in Debug mode) but also the RPM value doesn't even show up in the OSD anymore(See the pictures below) So it's not my equipment that's faulty, the esc telemetry just doesn't work.
[image]
[image]

## Steps to Reproduce
1. Enable ESC telemetry in BHelisuite(if esc doesn't have auto telemetry)
2. Flash FC with INav 2.5
3. Select Esc telemetry on spare UART
4. Set esc to Dshot1200
5. Drag Esc temperature and RPM to OSD

## Expected behavior
Esc temperature should output a numeric value
Esc RPM should display on the OSD and out put a numeric value

Matek F405-Wing Cli Dump all 
[PasteBin](https://pastebin.com/s1uh2Zw1)
Matek F765-Wing Cli Dump all
[PasteBin](https://pastebin.com/s4pSF05N)

- FC Board name and vendor: Matek F405-Wing and Matek F765-Wing

Mateyhv, 2020-06-09:
@Tduru , a known issue reported in #5818, to be solved in an upcoming release 2.5.1

giacomo892, 2020-06-10:
Fixed in #5825

closed 2020-06-10 by giacomo892