iNavFlight/inav #5844: Inav 2.5.0. VTX Tramp. Power levels limited / Pit mode not working

hrrr, 2020-06-11:
Setup:
Matek F411 Wing, and HGLRC micro VTX [https://www.banggood.com/HGLRC-FD-VTX-Micro-48CH-16X16mm-PIT25mW100mW200mW400mW-Switchabe-FPV-Transmitter-VTX-for-FPV-RC-Drone-p-1586870.html](url)

Diff: https://pastebin.com/QvVW3uun

## Current Behavior
With Inav 2.5,0 I can turn on Pit mode in OSD but it does not work (the VTX does not switch to pit mode)
Also the power levels are limited to 200mW max. Switching works fine, I just can not go to 400mW, 200mW ist the highest option than can be seen in OSD.

Pitmode and 400mW work both fine in Inav 2.4.0

digitalentity, 2020-06-11:
Likely this VTX doesn't report the max output power correctly. Tramp driver will respect the max power that VTX reports. We can probably fix this by providing a config option to override VTX capabilities.

Mateyhv, 2020-06-17:
@hrrr can you check if its working now on 2.5.1?

hrrr, 2020-06-18:
400mw is working fine with the same vtx in 2.5.1 using 

> vtx_max_power_override = 400

Pit mode is still not working

closed 2023-10-27 by DzikuVx