iNavFlight/inav #5847: Servos 5 and 6 not assigning correctly in 2.5.0

theproff777, 2020-06-11:
## Current Behavior
[image]
[image]

The screen dumps show visually what I see based on the stpes below.

## Steps to Reproduce
1. Select Flying Wing and load mixer
2. Deleted second motor because I don't have one
3. Added two servos and set first to servo 5 and second to servo 6 and both using RC Channel 5
4. Save and Reboot
5. Go to Outputs tabs to set values, but servo 5 is associated with S6 and servo 6 has no assignment
6. On the OSD I get a status message of 'PWM INIT ERROR'

## Expected behavior
I have upgraded from 2.3.3 to 2.5.0 and this is when the error on the OSD displayed.

This setup was working perfectly in 2.3.3.

## Additional context
# smix
smix 0 3 0 50 0 -1
smix 1 3 1 50 0 -1
smix 2 4 0 -50 0 -1
smix 3 4 1 50 0 -1
smix 4 5 8 100 0 -1
smix 5 6 8 100 0 -1

servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1540 -100
servo 4 1000 2000 1485 100
servo 5 1000 2000 1000 -100
servo 6 1100 2000 2000 -100

- FC Board name and vendor: Omnibus S4 Pro V3
- INAV version string:
# version
# INAV/OMNIBUSF4PRO_LEDSTRIPM5 2.5.0 Jun  7 2020 / 11:19:17 (694767938)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

tiriad, 2020-06-11:
There have been several changes since version 2.3.3. In your case I would advise you to use the target omnibusf4pro only.
 Current meter: omnibusf4pro
 Not current meter: omnibusf4v3

theproff777, 2020-06-11:
I quickly reflashed with omnibusf4pro and that solved the problem.

Now to wait for better weather so I can fly it.

Thanks 🥇

closed 2020-06-11 by theproff777