iNavFlight/inav #5849: Elevon deflections very small. Is this OK?

rsilk1949, 2020-06-12:
I believe I am ready for the maiden flight with the Dart 250g (flying weight 250 gm), running iNav 2.4 on a Matek 41S board.
When checking for correct deflection (nose up/down, roll right/left) the up/down directions look correct in manual, acro and angle mode.

However, when using the radio, the maximum deflection angle is very different with each of the 3 modes.
Manual looks like it is fine -- deflects around  +/- 5mm
Acro deflects around +- 2mm
Angle deflects around +- 1mm

Is this normal?

Also when moving the model around  (nose up/down, roll right/left) in angle mode, the max deflection is again only +- 1mm.

DzikuVx, 2020-06-12:
> with a confidence of 0.78. Please mark this comment with 👍 or 👎 to give

https://youtu.be/o2RwTeBbvos

rsilk1949, 2020-06-13:
Many tx for highlighting the video which explains everything. I will test after arming (prop off).

If these tips and tricks videos were placed in 1 or more youtube playlists and well advertised on the inav wiki, this knowledge base would be much more accessible.

rsilk1949, 2020-06-13:
Many tx for highlighting the video. I will test after arming -- props off ;)

closed 2020-06-14 by DzikuVx