iNavFlight/inav #5850: acc_lpf_hz default setting on 2.5.0

userod, 2020-06-12:
Hi all, not so much an issue per say but it seems like my 'acc_lpf_hz' set itself to 102 during set up after installing 2.5.0 FC firmware. How does this happen? (I hadn't flown the FC yet). seems like the default is down around 20..?

digitalentity, 2020-06-13:
You should flash will full erase

closed 2020-07-07 by DzikuVx

peteoz, 2021-07-23:
it's the same iNav bug that's causing "invalid settings" error, I don't think it's related to flash without full erase only, as I always do flash with full erase and still observed same issue.
my observation is this happens when CPU runs out of resources. some settings get corrupted and certain values are set to non-default values, in my case gyro_hardware_lpf, heading_hold_rate_limit and acc_lpf_hz got set to "101" and in case of gyro_hardware_lpf that's obviously invalid settings. it happened after I changed FC loop time from default 1kHz to 2kHz and after reboot iNav configurator became mostly unresponsive, I was still able to enter CLI and change the loop time back to default but the other settings were already corrupted.

joe8012, 2021-12-29:
This happened to me with inav 4.0.  acc_lpf_hz = 102

joe8012, 2021-12-29:
This showed up with Inav 4.0.  acc_lpf_hz = 102

MrD-RC, 2021-12-29:
It’s basically the same issue as this https://github.com/iNavFlight/inav/issues/7281

This will show you how to fix it https://inavfixedwinggroup.com/rc-news/inav-random-config-change-bug/

joe8012, 2021-12-29:
Thanks Darren.  I had three of the four parameters wrong.  No wonder I had such a hard time tuning my wing.