iNavFlight/inav #5852: Setting mixer to "Octo X8" on Matek F722SE disables ESCs

elstevedore, 2020-06-14:
## Current Behavior
I'm trying to run Inav 2.5.0 on an X8 Octocopter. Flight controller is Matek F722SE. ESCs are two Lumenier Elite 60a 4-in-1 ESCs. 

When I set the mixer to Octo X8, the ESCs stop working. It doesn't do the last two beeps, and the motors don't turn when I try to activate them manually. When it's in this state I also can't connect to the ESCs using BLHeli32 Suite. 

I've already enabled motor output, and if I set it to Quad X, then the first four motors work fine. 

**I have already tried setting the ESC protocol to Multishot**, since I've searched this board and read that Inav doesn't support DSHOT for more than 4 motors, but that doesn't solve it.  

I've tested it in Betaflight 4.1.7 and it works fine in Octo X8 mode with DSHOT 300. All 8 ESCs work normally under Betaflight.  

## Steps to Reproduce
1. Set mixer type to "Octo X8" 
2. Set ESC protocol to "Multishot" (I also tried DSHOT and Standard) 
3. Attempt to spin a motor in the output tab. 
4. Nothing happens

## Expected behavior
Expected all 8 ESCs to function normally and communicate with the flight controller. 

## Suggested solution(s)
Fix Octo X8 setting for Matek F722SE in Inav. 

---

- FC Board name and vendor: Matek F722SE
- INAV version string: Inav 2.5.0

Beardyblair, 2020-06-15:
Out of intrest what mode are you trying to arm in?

I also have this on an octa x running 2 x 4in1 esc.

elstevedore, 2020-06-15:
I've tried arming in angle and air modes, makes no difference. I don't
think the arming mode is a factor in my case at least, since the ESCs seem
to be completely non-functional with the mixer set to X8. As mentioned in
the OP, I can't spin motors in the output tab, or even connect to the ESCs
with BLHeli32 Suite when the mixer is set to X8.

danielmacy, 2021-08-28:
Can you explain what you mean by "You can always build the MATEKF722SE_8MOTOR target yourself."? A single sentence like that really isn't very helpful to me and likely most people who could use help in a similar situation. Do you mean you can go into INAV firmware flasher and select "MATEKF722SE_8MOTOR" from the drop down menu, "load online", and then "flash"? I'm not familiar with the lingo of "build the target". The link you provided simply shows 8 PWM are associated with the Matek F722SER, so I'm not sure what you're specifically trying to provide here for help. I was trying to provide others who run into the same issue I had, and struggled for hours to figure out, a solution that I eventually found worked for me. Please elaborate if you've got a better solution, or one that clarifies what I stated, cause I'm all ears (eyes of course). Can you please point me to a resource explaining how to do that?

Also, why is DSHOT not an option in INAV for an Octo X on Matek F722SE, when it works just fine in Betaflight?  It's a little confusing why motor 5 is the only one of all 8 motors that won't spin up when DSHOT is used, but will work fine if Multishot is selected.

stronnag, 2021-08-28:
> Can you explain what you mean by "You can always build the MATEKF722SE_8MOTOR target yourself."?

You clone the source repository and build the firmware yourself. There are comprehensive instructions in the [documentation](https://github.com/iNavFlight/inav/tree/master/docs/development).

Otherwise, as described in #7032, you can wait for the official MATEKF722SE_8MOTOR target to be provided in inav 3.1

> Also, why is DSHOT not an option in INAV for an Octo X on Matek F722SE, when it works just fine in Betaflight? It's a little confusing why motor 5 is the only one of all 8 motors that won't spin up when DSHOT is used, but will work fine if Multishot is selected.

See #7032 for MATEKSYS's opinion on this. You can always ask them.