iNavFlight/inav #5853: MATEK F411 unable to calibrate accelarometer after flashing 2.5

chilly24, 2020-06-14:
## Current Behavior
I flash from an older version to 2.5 and i am unable now to calibrate the FC accelarometer anymore.
I tried Version 2.5, 2.4, 2.2.1 still got the same results on windows and mac.
I flashed of course with full chip erase, the FC is mounted out of the wing and standing on a perfect leveled surface....

I am not sure if my gyro died.....

## Steps to Reproduce
1. MATEK F411 target
2. flash INAV 2.5 with full chip erase

## Suggested solution(s)
May its possible to check if its a dead gyro or its software related

## Additional context
https://pastebin.com/NkX9RYbc
https://streamable.com/fo2n4n -->I made i video exactly of the situation

---
# version
# INAV/MATEKF411 2.5.0 Jun  7 2020 / 11:48:17 (694767938)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

- FC Board name and vendor: MATEK F411 banggood
- INAV version string: 2.5.0

Mateyhv, 2020-06-14:
I have that board and have had no problems with acc, however what I use to do is just paste a 'diff all' saved from previous versions and it just works. Of course you have to modify the modes as per instructions in order to work on 2.5.0

chilly24, 2020-06-15:
Got it now, it was really a HW issue, the accelerometer was not working on z achsis as the super thin condensate was missing 😂
[image]

closed 2020-06-15 by chilly24