iNavFlight/inav #5854: Servo at Waypoint

fremebondo, 2020-06-14:
## Current Behavior
With logic conditions is possible to control servo based on navigation modes, but as far as I know, no logic condition can evaluate the actual mission leg number

## Desired Behavior
Specify the mission waypoint number in a logic condition.
I wish to use such behavior to autonomously detach a glider, tied to below my drone, in a specified position, then go back home and land, while I control the glider with a separate TX.

Is it possible in any way?
Thanks!

fremebondo, 2020-06-15:
I am trying adding 

`int8_t getWaypointIndex(void)
{
    return posControl.activeWaypointIndex;
}`

in navigation.c

and then using as LC input by adding a new enum value LOGIC_CONDITION_OPERAND_FLIGHT_ACTIVE_WP_INDEX
in logicFlightOperands_e, which triggers 

`case LOGIC_CONDITION_OPERAND_FLIGHT_ACTIVE_WP_INDEX:
            return constrain(getWaypointIndex(Z), 0, 32767);
            break;`

within case of static int logicConditionGetFlightOperandValue(int operand)

Am I doing it in the right way?

fremebondo, 2020-06-15:
Same as #5573 , please merge.

fremebondo, 2020-08-17:
The suggested solution works fine for my needs, but it is not so elegant indeed, for at least 2 reasons:
- You have to set the waypoint's id where you want to toggle logic conditions within logic condition itself, while it would be more correct to specify that at the waypoint. In this way if you edit a mission you set the servo switch point on a waypoint instead of having to edit the logic conditions too.
- posControl.activeWaypointIndex seems not to reset automatically when the craft is disarmed or the WP mode is disabled. Currently I unplug the battery after a flight, to start a new mission with an unset logic condition. I have to dive more deeply in the code to understand when activeWaypointIndex is actually set back to zero.

closed 2020-09-21 by DzikuVx

haiduc32, 2020-09-21:
this is very cool, but also raises a moral question: how easy you want to make it for someone to weaponize iNav?..

FelixvdDonk, 2020-09-21:
> 
> 
> this is very cool, but also raises a moral question: how easy you want to make it for someone to weaponize iNav?..

It is, good point. I felt like that was the major reason this feature wasn't in Inav sooner given the simplicity. But then again, Ardupilot has had this feature of performing actions (actuating servos) at waypoints for years. So the ease and capability has been there for a long time

haiduc32, 2020-09-22:
sure, but comparing Ardupilot with iNav is a bit unfair, since iNav can run on <25$ boards and IMHO is easier to set up.

FelixvdDonk, 2020-09-22:
> 
> 
> sure, but comparing Ardupilot with iNav is a bit unfair, since iNav can run on <25$ boards and IMHO is easier to set up.

So does ardupilot, you can run it on the omnibus f405 V3 and derived clones.

wittzer, 2023-07-22:
yeap ardupilot FC is not expensive

PR #6137 (2020-09-21): Ability to read current waypoint index as part of LC
This allows to trigger a servo output at waypoint with the help of Programming framework. See the following example

```
# If current waypoint is equal to previous waypoint index
logic 0 1 -1 1 5 7 2 29 0       
# Not LC#0
logic 1 1 -1 12 4 0 0 0 0
# If waypoint index changed and is WP mission, use this LC to react
logic 2 1 -1 7 2 23 4 1 0
# Store previous waypoint index
logic 31 1 -1 18 0 7 2 29 0
```

closes #5573
closes #5854